bk1285/rpi_wordclock #1: software Voraussetzungen

ghwc2015, 2015-01-17:
Hallo Bernd,

ich bin gerade dabei die Uhr nachzubauen.
Leider bekomme ich die Software noch nicht ans Laufen.

Die Abhängigkeiten scheinen über das hinauszugehen was auf der Anleitungsseite beschrieben ist.
die PIL habe ich noch installiert bekommen, aber jetzt verlangt das Programm "scipy" - und das scheint sehr aufwändig zu installieren zu sein.

Hast du evtl. noch eine genauere Anleitung bzw. Auflistung der Voraussetzungen?

Gruß
Timo

bk1285, 2015-01-17:
Hi Thimo,

cool. Du bist der erste, der sich daran macht, die wordclock basierend auf der vorgegebenen Anleitung zu realisieren...

Daher: Danke für' feedback ;)

Die Kommandos werde ich zur Anleitung hinzufügen: Die sollten's tun

sudo pip install scipy
sudo pip install pil

Grüße,
Bernd

closed 2015-01-17 by bk1285

ghwc2015, 2015-01-18:
Hi Bernd,

mit den unten angegebenen pip install Modulen hat es leider immer noch
nicht funktioniert.
Ich habe dann per apt-get python-scipy installiert und PIL durch pillow
ersetzt.
Danach ist das wordclock.py gelaufen.

auf der Seite "how to install" im Abschnitt "the wordclock software"
fehlt ein "h" in der URL

git clone https://git*h*ub.com/bk1285/rpi_wordclock.git

Die Ansteuerung per Levelshifter habe ich von der Adafruit Seite
genommen - da wäre eine Hinweis oder evtl. Link  auf der Hardware-Seite
hilfreich.
Für die Tasten gilt das auch - wenn keine Tastenschaltung angeschlossen
ist, meint das Programm ständig die Taste 17 gedrückt zu bekommen.

Für meine Wordclock kommt jetzt erst der aufwändige Teil des
mechanischen Aufbaus - derzeit habe ich noch mit dem Streifen auf der
Rolle getestet :)

Gruß
Timo

bk1285, 2015-01-18:
Hi Thimo,

cool. Danke für's feedback.

Bernd

ghwc2015, 2015-01-18:
nächstes Feedback.

nach dem heutigen git pull ist das Hauptprogramm nicht mehr gelaufen -
der Index des Plugin-Zählers scheint ein Problem zu haben (rot markiert)
- leider hab ich mit Python noch fast nichts gemacht.

Hab nur mal eine Ausgabe direkt vor der betroffenen Zeile
eingetragen(hier grün markiert)

Ich weiß nicht ob das hilft.
Manuell habe ich die Version vom 10.1. nochmal geladen - damit läuft es
wieder.

root@raspi05:/home/pi/rpi_wordclock# python wordclock-tt.py
Parsing /home/pi/rpi_wordclock/wordclock_config/wordclock_config.cfg
Importing plugin sunrise...
Importing plugin feed_parser...
Importing plugin time_as_words_german...
Importing plugin weather_forecast...
Importing plugin restart...
Importing plugin shutdown...
Importing plugin time_default...
  Selected time_default as default plugin
Index plugin_index7
Traceback (most recent call last):
  File "wordclock-tt.py", line 138, in <module>
    word_clock.run()
  File "wordclock-tt.py", line 107, in run
    self.runPlugin(self.default_plugin)
  File "wordclock-tt.py", line 82, in runPlugin
    print('Running plugin ' + self.plugins[plugin_index].name + '.')
IndexError: list index out of range
root@raspi05:/home/pi/rpi_wordclock#

Gruß
Timo

bk1285, 2015-01-18:
Hi Timo,

you catched the corner case... ;)

Habe mir das gerade nochmal angesehen. Sollte mit dem neusten Update behoben sein.

Grüße,
Bernd