bk1285/rpi_wordclock #4: wiring concept Grafik

ghwc2015, 2015-02-09:
Hi Bernd,

beim Zusammenbau bin ich mit der Grafik "wiring concept" nicht so ganz klargekommen.
Nummerierung der x- und y-Achse scheint mir nicht zu passen.

Ich habe für mein wiring Schema deine Grafik angepasst - vielleicht magst du das mit in die Doku aufnehmen.
Ich habe unten noch das Schema für setColorBy1DCoordinates eingefügt - diese Nummerierung ist hilfreich für die unterschiedlichen Fronten.

Gruß
Timo

[image]

bk1285, 2015-02-17:
Hi Timo,
thanks for reporting. That's actually a bug in the manual... ;/
Regards,
Bernd

bk1285, 2015-02-22:
Hi Timo,
danke fürs feedback. Habe ein skript hinzugefügt, welches es ermöglicht die wiring-layouts automatisch (und hoffentlich jetzt bug-frei) zu generieren... ;)
Bernd

closed 2015-02-22 by bk1285