bk1285/rpi_wordclock #9: Can't find fontdemo.py

Lexjans, 2015-04-19:
Hello I get a error by this command:
git clone https://gist.github.com/5488053.git
---------------------------------------------------------
pi@raspberrypi ~ $ git clone https://git.github.com/5488053.git
Cloning into '5488053'...
fatal: https://git.github.com/5488053.git/info/refs not found: did you run git update-server-info on the server?
---------------------------------------------------------
I update the server-info.
can you give me advise?

bk1285, 2015-04-19:
Hi Lexjans,

I cannot reproduce the issue. However, this post https://help.github.com/articles/https-cloning-errors/ indicates, that it might be related to your git-version?

I'm using version 1.8.3.2.

Best,
Bernd

Lexjans, 2015-04-19:
Hi Bernd,

Thank for answer my question.
I try it  again and now it works?????

I now have the software complete installed and can buy the LED's for the next step😀

Lex Janssen

closed 2015-04-19 by bk1285