bk1285/rpi_wordclock #10: Frage Verkabelung

rene100575, 2015-10-11:
Hallo,

ich will mal versuchen dein Super Projekt nachzubauen und habe 2 Fragen zur Verkabelung.
1) An welchem Pin hast du die Datenleitung der LEDs angeschlossen.
2) Wo schliesst man die Schalter an?

Vielen Dank und Viele Grüße
Rene

rene100575, 2015-10-11:
Hallo, 
so hab etwas rumprobiert - auf  GPIO 18 (GPIO_GEN1) bekomme ich die LEDs angesprochen. Bekomme aber immer die Melding "Pin 17 pressed" obwohl nix weiter angeschlossen ist...? Evtl. kann mir da jemand etwas auf die Sprünge helfen....

vg Rene

bk1285, 2015-10-11:
Hi Rene,

bzgl. der Tasten: Die Pins muessen dem config-file entsprechen:
https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_config/wordclock_config.example.cfg#L23

Bspw;
pin_button_left     = 17
pin_button_return   = 22
pin_button_right    = 24

Bzgl. "Pin 17 pressed": Das sollte sich aendern, wenn die Tasten angeschlossen sind.

Werde die Doku diesbezueglich nochmal updaten.

Hoffe das hilft. Ansonsten gerne das issue wieder oeffnen.

Gruesse,
Bernd

closed 2015-10-11 by bk1285