bk1285/rpi_wordclock #11: WS2812B mit WS2811 Controller

tooliload, 2015-10-18:
Hallo,
ich versuche seit einigen Tagen den WS2812B mit WS2811 Controller an einem Raspberry Pi 2 B laufen zu bekommen. Aktuell klappt es nicht. In Foren lese ich immer wieder das der WS2811 mit einem Pi nicht ansteuerbar ist. Ich habe die Installation gemäß deinem Howto vorgenommen, Fehler gab es keine. Ich habe verschiedene Raspian-Versionen probiert. Die Pythonscripte laufen und geben die Meldung aus das eine Taste gedrückt wurde, weil diese noch nicht vorhanden sind. Die LEDs rühren sich aber nicht. Hast du eine Idee?

Danke für das tolle Projekt.

stefancarlton, 2015-10-18:
Apologies for the English Reply...

I built this project using a Raspberry 2 and it worked successfully for me. I used an alternative libary, https://github.com/richardghirst/rpi_ws281x library which was built for Pi2. There is also the example script https://github.com/richardghirst/rpi_ws281x/blob/master/python/examples/strandtest.py to use.

I could not get the inverter to work though so wired mine up directly like Bernd's.

bk1285, 2015-10-24:
Hi @tooliload die alternativ-library nicht hilft, bitte melden... ;)
@stefancarlton thanks for the feedback! :)

closed 2015-10-24 by bk1285

tooliload, 2015-10-27:
Hallo, 

ich habe nun noch etwas neues probiert und bin zufällig auf etwas gestoßen. 

Der WS2812B mit WS2811 Controller aktualisiert sich bei mir nur, wenn ich ihn kurz vom Strom trenne und erneut anschließe. Hast du eine Idee warum? 

Die LED´s werden über den Python Script perfekt angesteuert, wenn ich dann das Netzteil vom WS2812B trenne und nach wenigen Sekunden wieder anschließe, leuchten die LED´s wieder aktuell. Wenn ich das Netzteil aber dauerhaft dran lasse, dann leuchten die LED´s zwar, aktualisieren sich aber nicht. -> Die Uhr bleibt stehen. (Gleiches auch bei dem Rainbow-Test, die LED´s aktualisieren sich nur, wenn der Strom kurz getrennt wird)

Ich habe nun erst einmal meinen alten Pi B+ am laufen. Die neue WS281x Lib für den Pi2 scheint zu funktionieren (zumindestens gab es keine Fehlermeldung), das die LED´s sich aber nicht aktualisieren, passiert auch hier.