bk1285/rpi_wordclock #12: Raspberry 2 support

kpankrath, 2015-10-19:
Hi,
I started building the wordclock as described in the documentation and used a Raspberry Pi 2.
It did not work. There was no error message, when running the wordclock.py or sandtest.py

After some research, I found out, that the ws281x package is not supporting Raspberry Pi 2 yet.
As I did the installation again on a raspberry pi 1, everything worked fine.

I would like to propose, that you highlight in the documentation, that the software is only running on Raspberry PI 1. 

This will reduce the time for others to search for mistakes...

Cheers from Berlin,
 Kai

stefancarlton, 2015-10-19:
The problem is not with this software, but rather the library it uses to communicate to the LED strip - if you use https://github.com/richardghirst/rpi_ws281x it works fine on the Raspberry Pi 2 (I built this using a Pi2 and this library).

bk1285, 2015-10-24:
@stefancarlton thanks for the feedback. Good to know, that it works with RPi2 as well.
@kpankrath please reopen this issue, if you still experience problems.

I furthermore added the point to the trouble-shooting section. I the origin library will not be updated soon, it might be a good thing, to move over to https://github.com/richardghirst/rpi_ws281x in general.

Regards,
Bernd

closed 2015-10-24 by bk1285