bk1285/rpi_wordclock #13: Ideen / Fragen

rene100575, 2015-10-27:
Hallo,

ich habe 2 Fragen:

1) Was habt ihr für einen Kleber für die LED Strips genommen - di sind ja so glatt das Heisskleber null hält :-(

2) Ich wollte das Projekt noch um 2 Funktionen erweitern:
 a) Temperatursensor/Luftfeuchtigkeitssensor und Anzeige der Werte.
 b) Abschalten der LEDs je nach Zeitplan oder wenn mein Handy nicht im WLAN ist. 
     (einfach zum Stromsparen)

Hat schon Jemand was in der Art versucht? Wenn ja würde ich mich über ein kurzes Feedback freuen.

Viele Grüße
Rene

bk1285, 2015-10-27:
Hi Rene,

zu 1): ich habe Heiskleber verwendet. Falls du eine bessere Alternative findest, melde dich gerne.

zu 2a): Hast du schon Hardware dafuer rausgesucht? Ich finde die Idee ganz nett und wuerde ggf. die gleiche Hardware kaufen und mitentwickeln... :)

zu 2b): Ich weiss von niemandem, der das bisher versucht hat. Klingt aber nett... :)

Beste Gruesse,
Bernd

rene100575, 2015-10-28:
Hallo,

zu 1) da es mit Heiskleber bei mir null gehalten hat habe ich einfach Silikon versendet - das Zeug womit man im BAD Fugen abdichtet - klebt wie Hölle trocknet leider sehr lange. 

zu 2a) ja, Hardwarekosten sind überschaubar ich werde einen DHT22/AM2302 verwenden ca. 5€. Hier gibt es einen schönen Blog dazu. 
http://www.sweetpi.de/blog/436/luftfeuchtigkeit-und-temperatur-mit-dem-raspberry-pi-messen 
Das einzige wo ich mal schauen muss wie ich die Werte dann mit den LEDs angezeigt bekomme - aber das ist für dich sicher ein leichtes ;-)

zu 2b) Ich dachte mir das so - ich pinge mein iPhone (im WLAN mit fester IP) alle 5 min an und wenn es 2 mal hintereinander nicht erreichbar ist - schalte ich alle LEDs aus. Wenn es wieder per ping erreichbar ist werden die LEDs wieder aktiviert. Außerdem wollte ich noch einen Zeitplan hinterlegen so das zwischen 12.00 Uhr und 05.30 Uhr die LEDs immer aus sind auch wenn das iPhone pingbar ist (dann schlafe ich und es liegt auf dem Nachttisch ;-))

VG Rene

bk1285, 2015-10-29:
Habe die Komponenten für 2a bestellt... Lieferzeit zwischen 7. und 27. November...

Bin gespannt... :)

rene100575, 2015-10-30:
Hi,

ja cool - sag bescheid wenn du sie da hast - meine Sachen sollen auch
nächste Woche kommen....

VG Rene

euchkatzl, 2015-11-13:
Hi,
erstmal vielen Dank für die Bereitstellung des Codes ! Sehr gute Arbeit.

Ich werde mir demnächst eine solche Uhr bauen und habe dazu zwei Fragen.

1. ) Da ich aus Bayern komme werde ich die Anzeige der Zeit ein wenig Anpassen. Den Code dazu kann ich bereitstellen sobald er fertig ist. Wie das geht habe ich bereits herausgefunden. Ich muss dazu nur eine neue time_german_bavaria.py Datei anlegen, meine Änderungen machen und diese dann als default Laden oder ?

2. ) Außerdem würde ich gerne die Helligkeit anpassen können. Da ich bereits eine WordClock besitze (nach Christians Anleitung auf Basis eines Arduinos) finde ich diese Funktion noch sinnvoll da die LED's in einen dunklen Raum doch schon stören können. Könnt ihr mir vll kurz einen Tipp geben wie ich das realisieren kann, sodass ich im normalen Modus die Farbe und Helligkeit regulieren kann ?
Die Helligkeit anzupassen dürfte nicht sehr schwer sein : 
https://forums.adafruit.com/viewtopic.php?f=47&t=41143

Vielen Dank im Voraus.

bk1285, 2015-11-13:
Hi,

zu 1.) stimmt. time_bavarian.py oder time_german_bavarian.py wuerden sich anbieten. Wenn das alles funktioniert, gerne auch hier sharen.... ;)

zu 2.) Das ist eine Sache, die ich auch schon ewig mal angehen wollte. Die neuste Version unterstuetzt nun das setzen der Helligkeit (brightness) im config-file.

Best!
Bernd

whybot, 2015-11-25:
Hello,

I hope English is ok, although I am planning on a French version of the clock! Can't quite believe I've stubbled across this, I'm at the point of written some far cruder python code to show the clock and set up my LEDS and cut the wooden frame, and just about to drill the holes when I found this! I've set up the LEDs and raspberry pi in practically the same way although I'm using an level converter, which I am not sure if you have do so. 
Any way the code is much more vast than i was intending but I suspect is more robust than what I have made up. I should be able to adapt it to my own set up, although my python coding knowledge is fairly new. The first issue I have is with the set of the LEDs, as I have only just found this I have wired the LEDS in the reverse direction so where your LEDS start 12345... mine is are in the reverse direction 54321 and so on. Can you provide some pointers so I can at least start to understand the code a little! 

I am also looking at running the clock in French if you have any pointers on how I go about adapting it to do so, the config file was not as intuitive as I thought! Also is it possible to by-pass the buttons for now to test the code, I'm not a the point of adding buttons in yet? e.g. just set default colour etc?

Thanks for posting this and providing inspiration to finish it. Attached an example of my work so far, before I have even seen this!

[image]

stefancarlton, 2015-11-25:
Hi,

I forked Bernd's code and removed the external buttons as I didn't need it for my version: https://github.com/stefancarlton/rpi_wordclock and moved the "minutes" to be within the matrix. I intend to update it to use websockets and a little website to replicate the buttons but haven't got around to it yet.

You can see my version over on http://clock.stefancarlton.net

To set the LEDs you simply need to create a language version and then specify the range of LED's, for example https://github.com/stefancarlton/rpi_wordclock/blob/master/wordclock_plugins/time_default/time_english.py. I would assume therefore it doesn't matter what the numbers are, you just need to specify the IDs which can be worked out using the strip test. In Python the "range" function just produces an array between the two numbers so if your example has odd sequence, just specify the ID's independently.

whybot, 2015-11-25:
Hi Stefan,

Thanks for the quick response! I think I understand the lang bit now, my original basic code before finding all this was as follows, for each word!!! I'll give your modified version version ago, I am undecided whether I wanted to add buttons/web control onto the clock yet, but that can come later! I assume for simply the clock I set all the "plug-ins" to false except time_default?

I need some more time to test out whats going on in the code but all your help is much appreciated!

#Il Est & HEURES
	strip.setPixelColor(0, color)	#I
	strip.setPixelColor(1, color)	#L	
	strip.setPixelColor(3, color)	#E
	strip.setPixelColor(4, color)	#S
	strip.setPixelColor(5, color)	#T
	strip.setPixelColor(60, color)	#H
	strip.setPixelColor(59, color)	#E	
	strip.setPixelColor(58, color)	#U
	strip.setPixelColor(57, color)	#R
	strip.setPixelColor(56, color)	#E
	strip.setPixelColor(55, color)	#S

stefancarlton, 2015-11-25:
my version has all the plugins disabled bar the time_default and the "start-up". The start-up is very actually nice and easy to understand how to get things to light up. 

https://github.com/stefancarlton/rpi_wordclock/blob/master/wordclock_plugins/startup/plugin.py
On startup the clock displays "It it Lina oclock" and that is done by specifying the individual LED's on line 24.

Taking your example above, you could create your language plugin such that
looks like 
IL = range(0,1)
ESTHE = range(3,5) + range(59,60)
URES = range(55,58)

And no worries on the replies - I loved making the clock and am grateful for Bernd making the code and docs available so happy to help others get the same enjoyment :)

bk1285, 2015-11-25:
Hi! Nice to see, how the "wordclock"-community is evolving...! :)

You're actually both right, that it might be a good thing to disable the buttons at the beginning. 

Concerning the wiring, I'd like to shortly point out the concept of the software:
It is provided in different layers. All plugins (the time, tetris,...) access hereby the top-layer of the coordinates for eased programming (top left (0,0), bottom right (10,9), alternatively 0-109): 
See for example https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_layouts/french_11x10/wiring_front_bernds_wiring.png

One layer below, are these coordinates mapped to the individual LED-wiring layout, using individual wiring-concepts, as you find here: https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_tools/wiring.py
"bernds wiring", maps the LED-strip to these coordinates:   0: min4, 1:min2, 2:(10,0), 3:(10,1), ...
Compare bernds wiring against christians or timos wiring...

These functions determine the course of the red line within the png-file... Here you need to model your individual one....

bk1285, 2015-11-25:
Hi @whybot  
... you might be lucky, and just choose "christians wiring" within the config file.

If it crashes (??) reduce the LED-count to
    self.LED_COUNT = self.WCA_WIDTH*self.WCA_HEIGHT
... and map the minutes all to 0 (since you do not have any leds for that...)

Keep us updated! Happy to get any news... ;)

bk1285, 2015-11-25:
After having this done, you need then to specify the individual words of your french layout, by creating a time_french.py in https://github.com/bk1285/rpi_wordclock/tree/master/wordclock_plugins/time_default as outlined by @stefancarlton

stefancarlton, 2015-11-25:
Aha, thats the piece of the wiring puzzle I forgot to mention - I simplified the grid generation for what I needed by hardcoding it to English (I ran out of time to go for niceties) https://github.com/stefancarlton/rpi_wordclock/blob/master/wordclock_tools/wiring.py#L70

SirHenryChicksatt, 2015-12-06:
Hallo, 
Eine Frage hätte ich. Hab mir die Matrix aus den LEDs zusammengelötet und das Testprogramm läuft auch perfekt durch. Wenn ich die Wordclock starte wird auch von meinem Layout die korrekte Zeit angezeigt, die LEDs wechseln nur ständig die Farbe. Die übrigen flackern ab und an kurz auf. Habt ihr eine Idee wie ich dieses Problem beheben kann?
Schon mal danke für einen eventuellen Tipp!

Grüße Chris

whybot, 2015-12-06:
Hi stefan,
I finally have got round to getting this adopted on my wiring layout but I have not been very successful. I've edited the wiring.py file what what I think is correct for the layout I have but it doesn't seem to be working very well, I've uploaded where I have got to on this file here
https://github.com/whybot/wordclockEdits/blob/master/wiring.py

In setting up the language file, does the range(0,3) for example mean 4 LEDs are lit 0, 1, 2, 3? As I am having issues getting that bit sorted out, although I am suspect of my wiring.py layout? Any help on this is much appreciated.

bk1285, 2015-12-12:
@rene100575 Temperatursensor ist angekommen.... Werde demnaechst hoffentlich die Zeit finden, das mal zu implementieren... :)

bk1285, 2015-12-27:
Hi @rene100575  habe den Temperatursensor integriert. Du darfst gerne testen und feedback geben! :)

To all other contributors of this issue: I close this for now, since it was initially intended to discuss the integration of a temperature sensor. ;)

In case you find other issues, open a new specific one ;)

Best,
Bernd

closed 2015-12-27 by bk1285

rene100575, 2015-12-27:
Hi,

Cool dann warst du schneller als ich ;-)
Ich werde es gerne testen befürchte aber das ich es erst im Jan schaffe...

Vg René