bk1285/rpi_wordclock #15: Sharing new Code

euchkatzl, 2015-12-09:
Hi,
As mentioned in Ideen/Fragen i've done a bavarian layout.
I've also added dynamic brightness switch (on pin 24) and a x-mas plugin (showing the text "Merry X-Mas).

How can i share my code ?
Here is a link to a patch file. Don't know if this is working.
https://www.dropbox.com/s/cspeqv0wfcxtm5b/patch.diff?dl=0

bk1285, 2015-12-12:
Hi!

I applied your patch file locally on my clock. There's really some cool stuff in there! Nice! I'll reiterate through it, if I find some more time and will then provide the code here! :)

Concerning your question, how to share code in general: The most convenient way is to

  * fork the latest master branch from this repository
  * apply your changes
  * create a pull-request (allowing me to automatically merge your changes)

... but the patchfile is fine as well... :)

Regards!
Bernd

Warent2454, 2015-12-17:
Hi,

I also did a git add and commit with several files needed for the layout i needed (10x10) in dutch. because the 11x11 already in the code does not work correctly. I hope you received it, else i will try again.

bk1285, 2015-12-21:
Hi @Warent2454  I did not receive anything... Maybe try to trigger a pull-request for your changes.

Regards,
Bernd

closed 2015-12-27 by bk1285