bk1285/rpi_wordclock #18: iOS App

euchkatzl, 2016-03-09:
I have created an experimental iOS App to Control the Wordclock from your smartphone. (Needs Xcode to Compile and Run)

Check it out here : 

https://github.com/euchkatzl/RaspberryWordclock-iOS

bk1285, 2016-03-12:
Hi @euchkatzl ,

That's a great thing! I like it a lot! I shortly inspected your repository and saw, that some diff needs to be applied to the wordclock installation in order to make it work with the app...

Maybe we should team up to modify the wordclock to be iOS-ready by default... ;)

I'd even consider one "global" solution, which considers android as well. I'll take a closer look, if I find the time: Probably not within the next two weeks... :/

Meanwhile I leave this issue open.

Best,
Bernd

euchkatzl, 2016-03-14:
I'm am already working on a better solution.

The best solution will be a simple web site hosted by the pi accessible through a normal web browser.

I have little experience with that but i will get this work i think.
I will post it here as soon as possible.

Would be great if you could apply this patch by default.

euchkatzl, 2016-03-16:
I have made the Web based solution.
Check it out : https://github.com/euchkatzl/RaspberryWordclock-RemoteControl.

My main Problem is that i cannot test all this stuff because the clock was a present for christmas and i have no possibility to give it a "live" try at any time. I will build mine in the course of this year.

Will be great if we get this working.

euchkatzl, 2016-10-02:
Hi Bernd,
finally  created my own clock and get the remote stuff working. (On all devices ;) )

Can you please check out the changes in my Repository and perhaps include it ? (https://github.com/euchkatzl/RaspberryWordclock-RemoteControl)

It is really cool playing tetris with controlling from your smartphone ;)

bk1285, 2016-10-02:
Awesome! I need to check this out!

I'll definitely add it to the repository! But tell me: Do your virtual GPIOs interfere with the real buttons? 

Regards!
Bernd

euchkatzl, 2016-10-02:
No. The real buttons work like before. They have the same functionality. 

Look at wordclock_interface.py file. 

It is just and "or" clause.

SimJaaenis, 2016-11-15:
Does it work at the moment? I tried it, but when I start the Webiopi-deamon I'm not able to use any button at the clock or at the web-interface.
I read, at the RaspberryWordclock-RemoteControl-project, you are using the Pins 7, 8 and 11. 
If you are referring on this wordclock-project, shouldn't it be Pins 11, 15 and 18?

euchkatzl, 2016-11-15:
Yes it works for me ! 
I have choosen GPIO's 7,8 and 11 (that is not the pin number! There is a difference. pin numbers for the Remote Control are 23,24 and 26). Of course you can use any other GPIO's. But then you have to modify the patch file and the index.html in my repo.
GPIO 18 you cannot use because it is used for the LED signal.

What did you do so far ? Did you see my Web Interface that i have written ? 
Have you applied the patch file to the wordclock repo ? It is not included by default

SimJaaenis, 2016-11-16:
Referring to this image [http://rpi-wordclock.readthedocs.io/en/latest/_images/wiring.png](url) I did my wiring.
Left button is at GPIO 17 = Pin 11, return button is at GPIO 22 = Pin 15 and  the right button is at GPIO 24 = Pin 18.
Just to understand the functionality: Do you use the same GPIO's like the original (hardware) buttons, or are the virtual "buttons" just referring to the same functionality as the hardware buttons?

Yesterday I saw your edited index.html with the "wordclock" link. Now I'm seeing the default page, without the link. I tried to many different things to get this to work, now its better if I clean install my wordclock.

euchkatzl, 2016-11-16:
The virtual buttons have only the same functionality as the original buttons.

But I think there is a wrong path file in the repo. 
I will check this as soon as i am at home today.

euchkatzl, 2016-11-16:
i created a fork with all the changes :
https://github.com/euchkatzl/rpi_wordclock (Commit : 0846e15d1ce63b189e55d58257882202ac61cf84)

euchkatzl, 2016-11-16:
You can simply clone ist to another folder. 
Then link the fontdemo.py and the this should work if you have configured webio-pi the right way.

I have tested this just right now ;)

bk1285, 2016-11-16:
Great, @euchkatzl,  Just merged it back: https://github.com/bk1285/rpi_wordclock/pull/39

bk1285, 2016-11-16:
@SimJaaenis can you confirm, if it works. If this is the case, I'll close the issue.
Bernd

SimJaaenis, 2016-11-17:
Short answer: Now It works! Thanks a lot!
Long answer: @euchkatzl In my case, I thought the htdocs-folder is the default of WebIOPi's-Webinterface, but WebIOPi doesn't use the "../htdocs/index.html" as default. After I tell WebIOPi in its config to do that, it worked.
I wondered a few times, why my browser didn't show any change of the index.html in 'htdocs'. After I changed the config-file to exactly that directory, it worked.  

`# Use doc-root to change default HTML and resource files location
doc-root = /home/pi/WebIOPi-0.7.1/htdocs`

Thanks a lot! Now I can play Tetris :D

A little improvement proposal:
an inactivity-time of the virtual buttons of 0.2 secs, after a virtual-button was pressed, would be good. Just like the hardware Buttons ;)

euchkatzl, 2016-11-17:
Great ! 
@SimJaaenis 
The timeout is 150 ms (look in the index.html). For the virtual buttons we have to simulate this.
Try setting it to 200 an report if it improves your stuff.

SimJaaenis, 2016-11-20:
I think the problem is my WiFi.
I changed the position and now it works better without any code changes.

bk1285, 2017-01-04:
Refering to https://github.com/bk1285/rpi_wordclock/issues/57

... whats the best way to go? If we find a fix for the current master branch, which leaves the virtual pins enabled, we're done at first place.
In the long run we could probably consider something more advanced...

bk1285, 2017-01-04:
Maybe we could also consider the work of @fecub which looks preety promissing...  https://github.com/fecub/rpi_wordclock/commits/remote_controller

;)

Tihmann, 2017-01-08:
How does @fecub modification work ? Is it setting up a server ? Tried to figure it out from the branch....

euchkatzl, 2017-01-09:
@Tihmann 
Can you tell me why Web IO Pi is not working anymore ?

bk1285, 2017-02-04:
Last weekend, I found the time to team up with @mrksngl, who is giving lectures in multi-threading, event-handling, etc. As a result of 14hrs of hacking, the wordclock will soon be ready to take not only the buttons as input anymore... Great work, @mrksngl :) 

Checkout https://github.com/bk1285/rpi_wordclock/tree/develop for further details or take a look at https://github.com/bk1285/rpi_wordclock/projects/2 for the further agenda.

euchkatzl, 2017-02-07:
Hi @bk1285 ,

That is great work ! Thank you for that. 
I've testet the socket connection with a litte Qt Programm. Seems to work fine ! 

But there is a little more work to do so that we can use it as Remote Control !

1. Add a command for Button press (e.g. sending  "BUTTON_PRESS" : 18)
    a.  I think that will not be the problem and very easy to implement
2. To work with any devices we need an web interface (like the page i have written)
    a. This page must be hosted on the pi to interact with the socket connection
    b. We need a web server on the pi 
         1. Host it directly via python code ( i dont know if this is possible)
         2. Host it via node (i think this is the easiest solution)
         3. Set up an apache Web Server for usage of php
3. Without a new Webinterface the socket connection could only be used directly in an app that we habe to write for ios and android. I can do this but then we need to upload it to the app stores so that anybody can use it ( could be a problem . Especially on ios)

So tell me what you think the best solution could be !? I can do the Web Interface work if you want to.

fecub, 2017-02-07:
Hi @bk1285,

thank you Bernd. Thank you for this awesome clock project.  
I wanted to report you, if I'm done with my work :) but you discovered me earlier :)

Ok what I have done:

-  as @Tihmann says. I set up an server and intergrated in the project.
- the remoting is working on main  menu, time_default and matrix plugins
- ... the other plugins need to modified for remoting
- I wanted the gpio button and remote controller things seperated. But this forcing me to add two or more lines of code

1. For example you need to pass the socket class on every plugin
2. you need to init a socket class on every plugin: `remote_event = wso.waitForEvent(0.5)`
3. ... and for the controlling `if (event == wci.button_left or remote_event == wso.button_left):`

- but with the seperation my plan was not only to remote the clock, also to monitor the hardware. and Controll the colors etc. 
- At the moment I'm writing an app with Qt. I have written an small client with python, too. 
Which i can controll it. 

...next step was to contributing on this project :)

mrksngl, 2017-02-07:
Hi @euchkatzl ,

1. I already implemented that, but did not push it so far. @bk1285 and I have to agree on some aspects beforehand.
2. I suppose it shouldn't be too hard to get this working. The new event interface accepts events from any thread, thus creating a thread hosting a webpage or doing IPC with an external webserver (i.e. outside the python script) should be managable. Anyway, I have to suggest that we open another Issue for this, as of right now, this won't be my main concern.
3. Android App is coming. This was the main reason of why we did that in the first place. Since I don't own an IPhone and don't know anything of iOS development (well, didn't know lot about Android development either until two weeks ago) I think there won't be an iOS app soon. So either someone else writes an iOS app or someone makes me an expensive present or the iOS users have to stick with the web interface. About the app store: An android developer account (needed for uploading) costs $25. Sorry, but I don't see a point in spending 25$ for one app, getting nothing in return (I don't plan that users have to pay for it -- to the contrary: I think I release the source on github soon). We can release the APK, so users won't have to compile the app theirselfs, but side load it directly. This will need some more steps to install the app instead of just loading it from the play store, but as the whole project is undoubtedly a hacking project, I think the average owner is keen enough to do it.

GPIO Py (I think this has something to do with those 'virtual gpios') is no longer working because I removed it. There are several reasons:
- using some otherwise unused GPIOs for Interprocess Communication is bad design. It's just misuse (even worse: of scarce resources).
- there are plenty of ways to do IPC otherwise, which can now be easily integrated into the new design of the event component.
- I removed _polling_ on GPIOs completely (which is also bad design) and replaced it by _waiting_ for edges on the pins (the hardware, kernel and the gpio library support this). This also reduces CPU load, as the script won't do anything at all most of the time now. In the old design it was asking about GPIO states most of the time, with little sleep in between.
- Waiting for edges is only possible on GPIO pins which are configured as INPUT pins. The virtual GPIOs were configured as OUTPUT pins.

mrksngl, 2017-02-07:
Little preview of my current work (yes, I emulated everything in here) (yes, GTK has some problems with umlauts): https://www-user.rhrk.uni-kl.de/~mengel/wclremote.webm

euchkatzl, 2017-02-07:
@bk1285 @mrksngl 

I totally agree on the "bad design" points ! The solution now is a much better approach !

For App Developments i can help you since i'm am an app developer (iOS and Android) in my main job ;)

Video of the App looks great ! How did you get that simulated wordclock interface ? Can you explain that (perhaps in another ticket)

For the Android app you are right ! You can distribute this without any App Store. But in iOS this is not possible.

So the best approach is to write a new Webinterface which works with both systems (and also Windows Phone). I can do this as soon as you publish your Android Code.
Moved discussion for this :
https://github.com/bk1285/rpi_wordclock/issues/59

mrksngl, 2017-02-07:
@euchkatzl 

Glad to hear we have an app expert:
1) I don't like GUI development at all (though I have to admit that android has some nice concepts and there is a really cool tool support in android studio)
2) I just started android development, so I read some guidelines (you know it: android is a beast) and realized the project "my own way". Especially data exchange between the network and the UI thread is maybe not the Android way.

The simulated wordclock interface is just a GTK window with a 11x10 table (plus the dots for the minutes), having 114 labels, each displaying one character. I replaced (it's a huge hack!) everything that has to do with the LED strip with GTK code recolouring the labels. The button events are generated by keyboard events. I could publish it, but only in a separate branch, because right now it breaks normal operation. This is another issue in refactoring the project.

euchkatzl, 2017-02-07:
1. Yes Android is very easy for GUI things. Much better than iOS.
2. I can tell you that when i got a look at your Code. For me it would be nice to get an idea how to talk to the socket ;)

The better approach would be using PyQt for the simulated GUI. So it works also on non Linux Systems (maybe). But would be great if you could publish it ;)
Thank you !

bk1285, 2017-11-03:
Hi all, just finished an experimental version of the webinterface: https://github.com/bk1285/rpi_wordclock/tree/feature/webpy
There are still some minor issues to fix, before I merge to develop, but maybe you want to give it a try (after installing web.py). Please note, that this currently requires the wordclock.py to be started within the directory, it's located in: /home/pi/rpi_wordclock.

Best,
Bernd

Sauerland1973, 2017-11-09:
Hallo Bernd,
habe dein Projekt noch ein wenig "verfeinert" ;-) Ich habe eine Matrix von 15 x 15 LED....
nun würde ich gerne auch das Web interface nutzen, allerdings bekomme ich diese Fehlermeldung:
File "/home/pi/rpi_wordclock/wordclock_interfaces/web_interface.py", line 1, in <module>
    import web
ImportError: No module named web

Was mache ich Falsch? 
LG aus dem Sauerland

euchkatzl, 2017-11-09:
Du musst WebPy installieren :
http://webpy.org/install

Sauerland1973, 2017-11-09:
Oh, das ging schnell, vielen Dank für die schnelle Antwort!
Ich werde es direkt mal ausprobieren ;-)
LG

closed 2018-01-04 by bk1285