bk1285/rpi_wordclock #21: Need help

christiankrueger16790, 2016-03-20:
Hi all,

Sorry, if i write down my comment into the wrong directory, but didn´t found a directory for questions.

I´m also want to build a RPi wordclock. I´ve got a rpi a+ model with ws2812 led controller.
Unfortunately, I'm still a little inexperienced in matters of led-controlling with a raspberry pi.
I´ve finished the hardware setup and installed the software like the description.
But when i´m start the wordclock software the leds, do nothing.
I´ve tried to controll the strips with the strandtest.py software, but the leds do also nothing.
Only two leds in the middle of the strips lights up.
I thought i´ve got a misstake in the wiring, but i´ve checked all the solder joints and i think there look good.
I tried to wired up the wordclock with a 74AHCT125 level converter and a 1N4001 diode like the description of (adafruit).

Can you give me an idea, what i also can check.
Sorry about my stupid questions and bad english!
I hope you can help me.

bk1285, 2016-03-20:
Hi @christiankrueger16790 

hm... What do you mean with ws2812 controller? You just need to connect a WS2812B led-stripe to the Raspberry.

Ideas from my side:
- ensure to have a WS2812B led strip (no WS2812: This is untested and possibly does not work)
- ensure to have the correct number of LEDs set to trigger (adopt the strandtest.py)
- ensure to have a power supply, which supplies enough power: With 10A at 5V, you're on the save side.

If you find the solution, maybe just post it here, so that others can benefit as well. Hope that helps. Otherwise, just reopen the issue.

Regards,
Bernd

closed 2016-03-20 by bk1285

christiankrueger16790, 2016-03-22:
Hi Bernd 

Sorry, but it is a ws2812B led stripe, not only ws2812.
I set the correct number of leds (114 counts).
I´ve got a power supply of 5V/10A, i think that´s ok.

Do you have another idea?

Can create and upload a zip file of your actual wordclock image with win32 disk manager, is it possible?
Because I want to ensure, that i don´t forgot any packages.

Regards 
Christian

bk1285, 2016-03-22:
Hm.... maybe try to go through the installation steps again: If you reinstall the packages, you might identify missing dependencies...

Regards,
Bernd