bk1285/rpi_wordclock #23: IR control

dgeordgy21, 2016-04-05:
Hi all!
Finialy I have attached the tree bouttons and they are working!
But it's not practical at all especially if the clock is mounted on a wall...
My idea is to control the clock with a IR remote control.
I'm not a programmer, so I found this code seems to do the job.
Do you think you can atapt it?
http://ozzmaker.com/how-to-control-the-gpio-on-a-raspberry-pi-with-an-ir-remote/

bk1285, 2016-04-07:
Hi @dgeordgy21 

I like the concept quite much. However I'd currently give #18 priority. If this work out well, you can control the clock with your smartphone or any other pc/laptop/tablet within your local network.

Regards,
Bernd

closed 2016-04-07 by bk1285

bk1285, 2016-10-02:
Hi @dgeordgy21 
@euchkatzl reports to have a working remote control as mentioned in #18 Maybe you want to give it a try. :)