bk1285/rpi_wordclock #24: stencil/clock face

whybot, 2016-04-12:
Hello, 
Are you able to provide link(s) to online vendors who can produce customised stencil/clock faces perhaps the online company used for this? -might be useful to add to the hardware list in the documentation.
Thanks :)

stefancarlton, 2016-04-12:
I used https://www.ponoko.com/ for my clock face in wood. They were very helpful in getting my layout in the correct format for their printers.

bk1285, 2016-04-13:
Thanks @whybot and @stefancarlton for the hints. I'll add it to the documentation.
Regards,
Bernd

bk1285, 2016-04-13:
... why not wait for possible further comments and ideas... ;)

i-pinz, 2016-07-31:
Hello Bernd,

What is the size of the wooden board you used? And the framing is done with alu-profiles or did you had a frame from an picture-frame. Your project looks nice, and I'm just collecting the pieces I need to make a shopping tour. What I miss is a some kind of measurement of the frame and the wooden board.

Thanks
pinz

bk1285, 2016-07-31:
Hi @i-pinz,

I used alu profiles. The wooden board needs to have the same size as your stancil. To make the alu-profiles fit, consider that they need to overlap at the corners. Hope, that helps?

Regards,
Bernd

i-pinz, 2016-07-31:
Hi Bernd,

I was looking around and I'm think I'm going to put it into a frame of an Swedish company that has a size of 50 by 50cm, I guess it should fit. It is also a rather bulky frame, so I hope I can fit the pi and the buttons into it.

Tomorrow I will grab on of the PIs I have and look the local shop of this Swedish company.

As I see it at the moment I have to create the stencil with the python program and get it done as written above a company like ponoko and put it onto the glass or in between. 

In the video and the pictures of the guide you use a rather thick wood, I guess I can also use a thinner one, or was it on purpose to get a better lighting of the characters?

Sorry to ask so many stupid questions but better to ask before starting than doing it once over again.

Regards
Pinz

bk1285, 2016-08-01:
Hi @i-pinz,

your concept with the frame of the swedish company sounds interesting! ;) Since it's a somehow new approach (compared to mine), it would be great, if you'd share a picture or two of the final result... :)

The thick wood improved indeed the illumination of the characters (ensuring a minimum distance between LED and plate).

Good luck and best regards,
Bernd

bk1285, 2016-08-01:
PS: If you rerun the python scripts with your own measures of the stancil, consider the correct spacing of the characters according to the LED-strip you use. If you choose the spacing to wide, your LED-strip might not be useable anymore... ;)

stefancarlton, 2016-08-01:
Hi pinz,

I did something similar with a wooden frame (albeit custom made) and you
can see the steps I took here http://clock.stefancarlton.net

One thing i learned is the importance of the diffuser material, the
distance between the led & the diffuser and the size of the letter cutouts.
The one i built is a bit glarey and doesn't glow like Bernd's does.

Remember as well you can get led strips with 60per metre so you can make
the clock much more dense.

Stefan

Tetrikus, 2016-08-02:
Hi all
I actually used the glass front of the Swedish wooden frame but have built a wood/alu frame around it since I felt like the frame is too deep. I reduced the depht of the clock to about 3cm and I guess the Ribba frame was a bit more.
Looks quite nice in my opinion but I will build a few more clock faces out of plexiglas in different colors and glue some magnets on it to make it easily replacable.

If you are interested in details and pictures:
_http://tetrikus.com/projects/portfolio/homemade-wordclock/_

But I also learned, that it's not a good idea to connect the strips one after another. After a few months of running fine, the clock broke and almost all of my LED strips where gone. I will now rebuild the clock from scratch and improve the power supply.

HeerNMeester, 2017-08-13:
I finished mine a few months ago but haven't shared my details. 
I have 2 front panels. An brushed aluminium and a Corten Steel one, which I let a few weeks outside to rust. I got them from Geers Cutting in Germany. (thickness 1 mm) You do have to provide them with the correct files. I tried to convert the .eps file from the stencil creator but that is not converted correctly by all programs I have tried. Eventually I manually adapted them in a CAD program. Not to difficult, but not straight forward. 
As for the font. Since it's laser cut, you need to have a special font that keeps the inside of some letters intact (A,O,R etc). I couldn't find a stencil font to my liking. I ended up adapting DIN regular with a font creator.
Oh tip: as a diffuser layer I used some greaseproof paper (the type you use in the oven to bake).

[image]

bk1285, 2017-08-31:
Woohoo! Looks great, @HeerNMeester !!

ruedli, 2017-12-22:
I noticed the STL files in the template directory and now I got a 3D printer (multi material) I was inspired to try this and take it a step further.

To summarize: 
I now use three different! templates with transparent, printed characters. I can shift between them with servo's and show "english" or "dutch" or "round dots". When no frame is used, you see 16x33 mm lighted rectangles, great for playing Tetris! The diffusers are printed as well: one layer of white (0.15mm) between transparant (2x 0.8mm on a stand).

The leds shine through transparent templates, everything is printed and the frames can slide.

I have finished the printed parts of the frame, remaining to do: hook up the pi, update the software to operate the 12 servo's, using a PCA9585, use mij wiring, use my dutch template with support for expessions for the minutes (het is zo / het is bijna / het was / het is nu / het is na / )

Enough to do therefore. I am willing to share my .stl files, they were made in fusion 360 from scratch.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

bk1285, 2017-12-22:
Hi @ruedli,

looks great, excellent work! Easiest way to integrate the stl is to create a pull request to "develop", which I can merge afterwards.

Best, 
Bernd

ruedli, 2017-12-22:
There is quite something to it, to reprint it you need over 400 .stl files. Many of them need to be sliced at the same time, because they come from "multi material"  model and need to be sliced in the proper position. The "B" for instance is a shape in the (black) template, filled with a transparent "B" and on top of the transparent "B" the inner two parts are modelled in black again. So... The character "B" is made of 3 bodies, not just one, as e.g. the character "T". In "Slic3r PE" this is a matter of drag / drop all .stl files that belong together in one go, then slice and print. This is needed for the templates (the 3 that I  prepared come as 4 quarter parts each, due to max size of my printer (205mm x 250mm) and the diffuser (this diffuser needs to be printed 110 times: one for each character position) 

An interested maker would probably need more slicer instructions, we will see. For now I attach all needed .stl files here with this post, with a "view" on these files in fusion 360. 

[image]

[image]

I will tune them further and when I am done create the "pull request" mentioned by Bernd. I also need to figure out how to update the software so that multiple (servo operated) templates can be configured without breaking it for the normal one template only.... check that the PI can do the 12 servo's through I2C and PCA9685 and need to study how software update is best done, suggestions are more than welcome here!  

Needless to say I am willing to help anyone courageous enough to try the same! In that case I will write more detailed instructions. So, if someone is interested in printing this 3D clock (it easily fits in the well known IKEA frame) let me know. 

I have printed mine on my Prusa i3 MKs2 Multi Material.

Have fun,
ruud

[WordClock_3d_print.zip](https://github.com/bk1285/rpi_wordclock/files/1582851/WordClock_3d_print.zip)

bk1285, 2017-12-22:
Hi @ruedli,

nice! Short video to see the clock in action is highly welcome!! ;)

Best,
Bernd

ruedli, 2017-12-22:
Yes, will do that when the servo's are working and when I got the software
all in one piece ;-)

ruedli, 2017-12-27:
I have created my own wiring sceme updating wiring.py, since I had glued my LED strips running n opposite direction. It now works and after small updates to reflect my specific templates the time displays correctly and and the plugins work.

I have the following questions: the numbers that are shown in the images of the of the "wordclock_layouts" (linke the wiring_front_bernds_wiring.pngand wiring_front_bernds_wiring.svg files) seem to be based on a horizontal strip arrangement, they do not correspond to the strip index and do not start e.g. with a "LED 0", what are they?

Can the stencil files and layout files be generated somehow or are they manually created/edited?

Is it meaningful to publish/provide updates to the stencils and wiring with the corresponding software updates? I have e.g. created an "english2" wiring and corresponding needed update in a (new) file called "time_english2.py" with provisioning line in "show_time" of the "plugin.py" of the time_default plugin, etc..

ruedli, 2018-01-02:
So... ITS WORKING... yesss, so pleased with the result...

Here is the video: https://youtu.be/XTHH30S8Aao

I have created a new "issue" to assist with merging the new plugin in the framework. I built in the support for servo switching in the existing classes for setting the time, this was needed to allow the core time software of the clock to switch between languages.

The change was done very cleanly, with hardly any impact to the existing classes. It now uses a list of classes for each language (and only the relevant ones) that needs to be support, rather then only one loaded class. 

This was done by support for reading a comma separated list of languages, rather then "only one" language. A new plugin can be used to set the "active_stencil". The servos are configured in there own configuration section called "servos".

aupp, 2023-08-12:
> I finished mine a few months ago but haven't shared my details.
>
> I have 2 front panels. An brushed aluminium and a Corten Steel one, which I let a few weeks outside to rust. I got them from Geers [...]

A long time since you did this, but would it be possible to share the font yiu created? If you still have it? 😊
The clocks looks awesome btw! How did the rusty corten one end up?

ruedli, 2023-08-13:
Well.... because I 3D printed it from plastic, I printed the characters from transparent plastic in the black plastic that surrounded the characters. For the diffuser layer I used one layer of white filament :-)

The font is plain old Arial

HeerNMeester, 2023-08-13:
@ruedli I think he is asking for my font ;-)
@aupp I just made a pull request with the stencil font I made. Luckily I had it somewhere on an old cloud account...
I prefer the brushed RVS over the Corten steel plate. It turned out quitte good, but it doesn't fit in our livingroom. I did made a few other ones, like this one in Bamboo. Letters are lasercut, but because of the thickness of the plate (5mm) the view from an angle is bad (you are looking into the sides of the letters and the light is not coming trough. So I filled them with an epoxy resin with a diffuser additive. The pouring was easy, only the sanding afterwards was quite a job! 
[image]

ruedli, 2023-08-13:
Ja, denk ik ook :-)

ruedli, 2023-08-13:
If you fill it up with resin, you can temporarily glue "loose" centre parts
of characters to a flat surface together with the stencil and then pour the
resin. Anyone tried that?

aupp, 2023-08-13:
Yes, you're right, the DIN-font was what I was hoping for. Very nice
😊Thanks!

The bamboo looks really cool too! 👍