bk1285/rpi_wordclock #25: christians_wiring nutzen

tooliload, 2016-05-14:
Hallo, 

nach dem ich nun einen neuen LED Streifen bestellt habe, geht das Projekt auch bei mir weiter. Tatsächlich konnte mein alter LED Streifen mit genau den selben LED´s nicht gesteuert werden. - Defekt am LED Streifen .... 

Der neue LED Streifen klappt nun wunderbar und Probelauf mit /root/rpi_ws281x/python/examples/strandtest.py läuft hervorragend. 

Ich habe die Uhr nach dem Layout "christians_wiring" angefertigt. 
Aktuell stehe ich jedoch vor dem Problem das die Ausgabe aber keine richtigen Wörter anzeigt. Es fehlt zum Beispiel "Es ist" und aus dem Wort "Uhr" wird "Muhr". Dies LED´s leuchten wild durcheinander. 

Das Layout, die Sprache und die Maße meiner Uhr habe ich natürlich eingestellt: 

/root/rpi_wordclock/wordclock_config/wordclock_config.cfg

`
[wordclock_display]
wiring_layout = christians_wiring

[stancil_parameter]
height=450
width=450
minute_margin=30
minute_diameter=5
wca_height=300
wca_width=295
font_type=Arial
font_size=20
language=german

[plugin_time_default]
activate = True
language = german

[plugin_time_as_words_german]
activate = False
`

In wie weit müssen die Maße der Uhr stimmen? - Muss ich noch zustzlich was beachten, dass das Layout von Christian geladen wird? Zugegeben: Wenn ich den Parameter auf das Layout von Bernd ändere, sieht die Ausgabe der Uhr komplett anders aus, aber immer noch eher Buchstaben-Salat als Uhr.

Wäre klasse, wenn ihr noch eine Idee habt. 

Danke vorab.

tooliload, 2016-05-14:
[image]

Dieses Bild sollte eigentlich die Uhrzeit 11:12 Uhr zeigen. 

Rückseite / Verkabelung: 
Die erste LED die vom Pi angesteuert wird, ist die 4 (unten links), dann die 3 (unten rechts), dann die 2 (oben links), die 1 (oben rechts) und anschließend geht die 1 zum ersten oberen Streifen auf die linke Seite. - So habe ich es aus der Grafik "german back-view christians_wireing" ausgelesen.

tooliload, 2016-05-14:
Bastelstunde beendet. So sieht mein Wiring Layout aus.

`
class tinos_wiring:

    def __init__(self, WCA_WIDTH, WCA_HEIGHT):
        self.WCA_WIDTH   = WCA_WIDTH
        self.WCA_HEIGHT  = WCA_HEIGHT
        self.LED_COUNT   = self.WCA_WIDTH*self.WCA_HEIGHT+4

        print("Show Leds:")
        print(self.LED_COUNT)

        print "Height / Width: ", self.WCA_HEIGHT, " / ", self.WCA_WIDTH

    def getStripIndexFrom2D(self, x, y):

        if y%2 == 0:
            pos = (self.WCA_WIDTH*y)+(self.WCA_WIDTH-x)+3
        else:
            pos = (self.WCA_HEIGHT*y)+(y+4)+x

        return pos

    def mapMinutes(self, min):

        if min == 1:
            return 3
        elif min == 2:
            return 2
        elif min == 3:
            return 1
        elif min == 4:
            return 0
        else:
            print('WARNING: Out of range, when mapping minutes...')
            print(min)
            return 0

`

bk1285, 2016-06-09:
Hi @tooliload 
super, dass das geklappt hat. D.h. es gab Fehler bei Christians wiring? Da du das ja lösen konntest, schließe ich das issue...

Beste Grüße,
Bernd

closed 2016-06-09 by bk1285