bk1285/rpi_wordclock #27: Ambient Light Sensor

nathangoss, 2016-07-23:
Hi Guys!

Am looking to add an ambient light sensor to the RPI_Wordclock.

I can see a fair bit of brightness-related code in wordclock_plugins/time_default/plugin.py however can't seem to work out the best place to incorporate changes based on readings from a light sensor.

Has anybody thought about this?

Thanks,

Nathan

bk1285, 2016-07-24:
Hi @nathangoss 

nice idea to integrate that!

Querry the brightness from your sensor within the while-loop:
https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_plugins/time_default/plugin.py#L79

... and update the brighness value:
wcd.setBrightness(your_value_within_the_range_0_to_255)

before the  wordclock display is updated in line 89:
https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_plugins/time_default/plugin.py#L89

Which light sensor do you have in mind? Would be great to hear about your progress! :)

Regards,
Bernd

euchkatzl, 2016-10-06:
Hi @bk1285 @nathangoss ,
i will do this this within the next few weeks.

I will use the following light sensor:
http://www.ebay.de/itm/Lichtsensor-Fotowiderstand-/291751650665?hash=item43edc0e569:g:ugkAAOSwMNxXZKqY

I keep you informed about the results.

euchkatzl, 2016-11-16:
I got things working with this sensor:
https://www.adafruit.com/product/439

it pretty easy because there are libraries which can be used. Of course the normal brightness stuff has then to be deactivated.

I will try to integrate that in the wordclock config and then publish the code.

bk1285, 2016-11-16:
... would be great to see your solution, @euchkatzl ;)

euchkatzl, 2016-12-20:
@bk1285 
f6a82d5b1428beb95e0b0bb857dce6a46558f2da
Finally i got it ;)

This Library must be uses to read out the sensor :
https://github.com/adafruit/Adafruit_Python_GPIO

HeerNMeester, 2017-08-13:
I got your code working with the same sensor @euchkatzl. However with the typewriter plug-in, every time the ammount of LUX changes, the time is written again. Creating a very 'busy' clock. This forces me to disable the typewriter plug-in.

spintike, 2018-06-15:
@euchkatzl Thanks a lot for the input you've given so far. I tried to implement your solution into my working example (by copying 'plugin.py' from your commit over my already working one) but I'm always receiving an error when running the script:

[image]

Any idea how I can debug the time_default plugin or what I'm doing wrong?

FrankX0, 2018-10-31:
Hi Bernd,

Did you consider to put the ambient light sensor support in the source?
I am using the implementation of @phenze which works fine for the wordclock plugin. But the other plugins do not use this brightness settings. And I'm using a different sensor (BH1750), which means I had to adapt the code a bit.
So an ambient sensor implementation which

1. Works for all plugins, and which
2. Supports a configurable sensor

Would be very welcome.

bk1285, 2018-11-01:
Strange, that the other plugins do not adopt this. I just ordered the sensor and will look into it.
... keep you updated, @FrankX0 @phenze

bk1285, 2019-01-01:
Hi all,

before starting the integration, I'd like to clarify the requirements for the light sensor.

So far I have on my list...:
* Setting of global brightness -- independent from the chosen plugin
* Handling of the measured brightness
  * Since we can also set the brightness via API, I'm considering to introduce an "auto-brightness" mode, which relies on the measured brightness from the sensor.
  * Via API, one can enable or disable the "auto-brightness" mode, in order to manually set a brightness value
* Additionally, I plan to integrate @cibears PR to optionally specify a period of time per day, where the clock uses a predefined low brightness -- independent from the light sensor values: E.g. standby at night...

Did I miss further aspects?

Best,
Bernd

@FrankX0 @phenze @cibear

Arjan444, 2020-02-27:
Hi  Bernd,

Love your Clock. Working on my 5th. 

The first 4 all have the TSL 2561 ambient light sensor (thanks Phenze). Did anyone already produce a version (a Beta maybe) of an integrated version of the latest software including this (or similar) ambient sensor?

Very much appreciated. Best, Arjan

@bk1285 @FrankX0 @phenze

bk1285, 2020-02-27:
Hi @Arjan444 
good to hear, that you like the clock! :)

Recently, I refactored the wordclock display. Changes are currently on develop branch. Especially the way of handling the brightness was [reworked](https://github.com/bk1285/rpi_wordclock/blob/develop/wordclock_tools/wordclock_display.py#L268).

The remaining work is to read out the current ambient light intensity and set the brightness accordingly.

Best,
Bernd

phenze, 2020-02-28:
Hi
my Master is on the actual Master of Bernd. Ambient Light Sensor should work.

Arjan444, 2020-02-28:
Ok. Thanks. I use the TSL 2591. Maybe it requires different library.

Arjan444, 2020-03-08:
Hi @bk1285 and @phenze, 

In both of your Masters my TSL2591 ambient light sensor does not do anything. I have it working in my other clocks but with the new wordclock display code and all I'm not sure how to fix it...I just hope one of you has a version in which it does work.....

Thank you, 

Arjan444

Arjan444, 2020-03-08:
@bk1285 and @phenze,

Could it be that the wordclcok code is written in Python 2.7 and that Adafruit GPIO libraries are V3 only? 

Thanks

bk1285, 2022-09-01:
Please take a look at the latest documentation to see how the ambient light sensor can be integrated: https://rpi-wordclock.readthedocs.io/en/main/doc_software_setup.html#optional-brightness-sensor

closed 2022-09-01 by bk1285