bk1285/rpi_wordclock #28: not functioning correctly

ELViTO12, 2016-08-25:
Hello, 
I have some troubles getting the clock to work. I connected all the cables to my rpi2b following bernds wiring and with the test function of the rpi_ws281x package i tested all the leds and they work. However when i start the rpi_wordclock all the leds randomly start flashing but it does not show the time. I installed raspbian and all of the packages twice and allso changed the rpi_ws281x library to  'Richards' library but that didnt work so i changed back to the 'original' jgarff library. Any ideas what i can change next?  Is there a way to start the plugins from the terminal  because i also dont understand what the buttons do. Thank you for your help!

tooliload, 2016-08-25:
Hi have you setup the right wirering in settings? What wirering layout use you?

ELViTO12, 2016-08-25:
I am using bernds wiring which is also default in the wordclock config so i did not have to change that. 
I saw in the specifications of rpi_ws281x : 'Converting the output from a Raspberry Pi GPIO/PWM to a higher voltage through a level shifter is required.'  I however did not see it in the layout of the tutorial nor in the video of the wiring so i also didnt do it,or maybe i missed it. Maybe that is my error?

bk1285, 2016-08-28:
Hi @ELViTO12, 
thanks for reporting. That's very strange! Sebastian, a friend of me, ran into the very same problem few days ago.

First, I thought about blacklisting the audio-kernel (as described here https://github.com/jgarff/rpi_ws281x/commit/27229b24028082b2ef9314c6247f621af296486d )
But since the LEDs work with the rpi_ws281x-package, this should not be the point.

I'll meet with Sebastian throughout this week to investigate. I'll keep you updated! :)

Regards,
Bernd

bk1285, 2016-08-29:
Hi @ELViTO12,

in our case, it's about blacklisting the audio-kernel... Can you check this: http://www.instructables.com/id/Disable-the-Built-in-Sound-Card-of-Raspberry-Pi/ and confirm? If this resolves your issue as well, I'll add it to the docs. Please confirm.

Regards,
Bernd

ELViTO12, 2016-08-29:
Hi Bernd,
 
Thank you very much for responding and finding a possible solution! I will follow the steps of the tutorial this week and will keep you posted.

Regards

ELViTO12, 2016-08-30:
I just implemented the solution and it works like a charm! Thank you very much for your advice :smile: 
 I also wanted to note that I have installed the rpi_ws281x package of jgarff on my raspberry pi 2 and it works. The recommended "Richards library" does not work.

bk1285, 2016-09-10:
@ELViTO12 @tooliload 

thanks for your reports/advices. I updated the docs accordingly: https://github.com/bk1285/rpi_wordclock/commit/cb47c9d8fd06cefb08efb13101c305efcdc691e7

Regards! :)

closed 2016-09-10 by bk1285