bk1285/rpi_wordclock #29: Dimensions of the LED strip

protectionofficer, 2016-08-28:
Hi,

Just about to start my adventure into your WordClock and would like to asked, which LED strip I should buy as they are so many on the market.

The options that are available are:
30 LEDs/m 
60 LEDs/m
74 LEDs/m
96 LEDs/m

I have found a reasonable eBayer in the UK : 

http://www.ebay.co.uk/itm/1-5-30-60-144-leds-m-WS2812B-5050-RGB-IP20-IP67-5V-Addressable-LED-Light-Strip-/112084027623?var=&hash=item1a18ba88e7:m:mjGK79p8uQg7zs-l8orE1hw

Also I have purchased the new Raspberry Pi 3 with the inbuilt WIFI, will the software still work on this model or should I purchase the Pi2?

Thank you for all your instructions, cant wait to get started.

Darren

bk1285, 2016-08-28:
Hi @protectionofficer consider, that you need at least 114 LEDs to use the wordclock. So consider for example this version here: 5m with 150 LEDS: http://astore.amazon.co.uk/araspbepiba0c-21

I didn't try with the RPi3 yet, but since jgarffs library should be compatible, it *should* work out. https://github.com/jgarff/rpi_ws281x 

Maybe give it a try and report back, if it works out with the RPi3. Would be great!

Regards,
Bernd

protectionofficer, 2016-08-29:
Thanks for quick response, will get back to as soon as I can with the update. 
Thanks again for all your knowledge. 

Darren

closed 2016-08-29 by bk1285