bk1285/rpi_wordclock #30: Angaben zur Holzplatte

Tihmann, 2016-09-06:
Hallo, erstmal vielen Dank für das coole Projekt. Möchte jetzt auch mein eigenes Projekt starten. Ich habe in der Beschreibung keine Angaben zur verwendeten Holzplatte gefunden. Vermute es könnte eine MDF Platte sein. Welche Abmessungen, welche Stärke ? Und welchen Durchmesser haben die Bohrlöcher bei Eurem verwendeten 30 LED/m LED Strip ?

bk1285, 2016-09-10:
Hi @Tihmann 

Bei mir waren das 16mm MDF mit 20mm Durchmesser der Bohrlöcher. Das variiert jedoch mit dem Layout deiner Frontplatte: Je nachdem wie der Abstand der Buchstaben ist, kann man auch größere Lochdurchmesser wählen.

Cheers,
Bernd

closed 2016-09-11 by bk1285

Tihmann, 2016-10-01:
Hallo,

ich habe doch noch ein paar Fragen, wie groß war der Abstand zwischen den LEDs auf Deinem LED Strip und woher hattest Du das Layout der Frontplatte ? Ist es das aus dem Github ? Wo hast Du die Glasplatte drucken lassen ?