bk1285/rpi_wordclock #31: Design Frontplatte

Tihmann, 2016-10-01:
Hallo,

ich habe doch noch ein paar Fragen, wie groß war der Abstand zwischen den LEDs auf Deinem LED Strip und woher hattest Du das Layout der Frontplatte ? Ist es das aus dem Github ? Wo hast Du die Glasplatte drucken lassen ?

Vielen Dank.

Liebe Grüße,

T. Ihmann

ulab, 2016-10-01:
Das Layout findest Du in https://github.com/bk1285/rpi_wordclock/tree/master/wordclock_layouts/german_11x10 (Ich vermute mal, Du brauchst die deutsche Version :)).

Das SVG-Layout kann man mit einem Schneidplotter aus einer selbstklebenden Vinyl-Folie ausschneiden lassen. Folienbeschriftungs-Dienste gibt es normalerweise in jeder Stadt oder auch online.

Tihmann, 2016-10-01:
Vielen Dank. Wie war denn der Abstand der LED auf dem Strip?

ulab, 2016-10-02:
In der Doku gibt es einen Link auf einen Amazon-Store für die Komponenten. Dort wird das folgende Produkt benutzt: https://www.amazon.de/WS2812-integrierten-Controller-Anschlusskabel-wasserdicht/dp/B00GXGOJLQ/ref=sr_1_1?ie=UTF8&qid=1475392632&sr=8-1&keywords=B00GXGOJLQ. Der Abstand wird dort Mitte-Mitte mit 33 mm angegeben. (In einem Kommentar, in der Beschreibung steht was von 17 mm...?!?)

euchkatzl, 2016-10-02:
Schau mal hier :
http://www.mikrocontroller.net/articles/Word_Clock

Da findet mann eigentlich alles was gebraucht wird.
Auch eine dfx Datei falls du dir ne Edelstahl Front lasern lassen willst .

euchkatzl, 2016-10-02:
Noch ne gute Quelle
http://diskussion.christians-bastel-laden.de/

bk1285, 2016-10-02:
@euchkatzl Done! :)

closed 2016-11-16 by bk1285