bk1285/rpi_wordclock #33: Couple of Questions

ebeng, 2016-10-30:
Hi Bernd,

I would like to thank you very much for putting this Worclock into the world of living :)
Last couple of months i've made my own clock.
The layout of the stencil i used is from here: https://cdn.instructables.com/F1D/NLCG/HYPNVXKB/F1DNLCGHYPNVXKB.LARGE.jpg
The clock is including the wood etc 65cmx65cm.

There is for now a couple of things i would like to ask you now.

My Leds are beginning from the right-below corner and goes to the left-below corner. 
At last it ends at the top row from left to go to the top right. When i would create a layout with the py file, how do i need to edit how the way of these leds are going. Also i dont have 4 LED's in the corner. You have the first 2 Led's also configured as 2 stips in the corner. So how do i get these out, and how do i configure my own stencil? 
I could rotate my glass to have it like your way, but the wiring etc would end in the left side for the powering etc, and also in the top i am going to make a brightness controller :)

I saw also that i need a make my own "time_as_words_english" for my own stencil, so thats also a thing in my mind.

Further, in the wordclock.config.cfg i put english also in it, and wrote all of the characters in a matrix.

Do i forget something else now?

ebeng, 2016-11-16:
anyone?

euchkatzl, 2016-11-16:
Because you have an completely different layout only @bk1285  can answer this question.
The normal range goes vertical not horizontal. That is the reason why everything is weird shown.

bk1285, 2016-11-16:
Hi @ebeng,

ok... I'll try with some inline comments. Note, however, that there's a significant amount of work ahead for you! There are some tricky parts, where I'm not sure, whether we get smoothly trough...

Consider furthermore, to push your changes, so that we can have a look at them: e.g. here: https://github.com/ebeng/rpi_wordclock

Regards,
Bernd

> Hi Bernd,
> 
> I would like to thank you very much for putting this Worclock into the world of living :)

Thanks :)

> Last couple of months i've made my own clock.
> The layout of the stencil i used is from here: https://cdn.instructables.com/F1D/NLCG/HYPNVXKB/F1DNLCGHYPNVXKB.LARGE.jpg
> The clock is including the wood [...]

Please clarify: top left, ...: Is this related to the front or backside?

> I could rotate my glass to have it like your way, but the wiring etc would end in the left side for the powering etc, and also in the top i am going to make a brightness controller :)
> 
> I saw also that i need a make my own "time_as_words_english" for my own stencil, so thats also a thing in my mind.

Yes

> 
> Further, in the wordclock.config.cfg i put english also in it, and wrote all of the characters in a matrix.
> 
> Do i forget something else now?
> 
> EDIT:
> 
> wordclock_plugin/time_matrix --> added my own "time_english" in the plugin.py
> wordclock_plugin/time_default --> added my own time_english.py , but i need to rerange the text with my own stencil
> wordclock_plugins/time_as_words_english --> made this one, according to the Dutch one
> 
> I can see the startup message, but it is so weird show... it goes from the below to the top, but you can see that it is not fitting my clock....
> 
> EDIT 2:
> in the create_layout.py i found out that you add +4 in the led_count on line 126, i took this out, but no luck still with the startup message. Also i made the font size smaller in the wordclock_config.cfg and run the "create_layout -c file" with it.
> 
> I seriously am at the place, where i dont know where to look now...
> 
> EDIT 3:
> f1dnlcghypnvxkb large
> 
> I edited the time_english.py in the time_default, to these values.
> Example : range 240-242 and range 243-245
> range 131-138

bk1285, 2016-11-16:
... please also have a look at the branches
https://github.com/bk1285/rpi_wordclock/tree/patch-1
https://github.com/bk1285/rpi_wordclock/tree/patch-2
https://github.com/bk1285/rpi_wordclock/tree/patch-3

... might give you at least a starting point.

bk1285, 2016-12-25:
Please reopen in case of any further questions/issues.

closed 2016-12-25 by bk1285