bk1285/rpi_wordclock #34: Uhrzeit wird angezeigt, zusätzlich jedoch weitere Buchstaben

Nthl25, 2016-11-06:
Hallo,

ich habe gestern meine WordClock fertig gestellt und hatte kurzfristig die korrekte Uhrzeit in weißer Schrift auf meiner Uhr.

Wenige Sekunden später wechselte die Farbe zu blau und seither leuchten zusätzlich noch einige Buchstaben in weiß.

Hat jemand von euch eine Ahnung, woran das liegen könnte?
Die Blaue Farbe ist übrigens auch nicht durchgängig gleich hell/dunkel.

Grüße

euchkatzl, 2016-11-06:
Hast du schon die Punkte von diesem Ticket überprüft ? 
https://github.com/bk1285/rpi_wordclock/issues/28

Der audio kernel war bei mir das problem.

Zusätzlich bitte mal den strandtest.py im rpi_wx /examples/python ordner ausführen und testen ob der richtig läuft.

Nthl25, 2016-11-07:
Den Eintrag hatte ich ursprünglich schon gemacht.
Nachdem ich die Datei nochmal angelegt habe funktioniert es (Rechtschreibfehler waren keine drin...).

Danke dir für deine Hilfe!

closed 2016-11-07 by bk1285