bk1285/rpi_wordclock #35: Word clock displaying wrong colour in the plugins

protectionofficer, 2016-11-07:
Hello Bernd,

Finally finished the WordClock (English version) and noticed that my colours in various Plugins are the wrong colours compared to your video.

The main visual difference is with the Matrix Plugin, the colour is a bright red instead of green.
I have checked the 'time_matrix/plugin.py' file and it clearly shows green as the HTML code?

I can only guess the hardware or software has the wrong RGB calibration?

Hope you can help?

Kindest regards
Darren

bk1285, 2016-11-07:
Hi Darren,

can you try to replace line 29:
`self.strip = Adafruit_NeoPixel(self.wcl.LED_COUNT, self.wcl.LED_PIN, self.wcl.LED_FREQ_HZ, self.wcl.LED_DMA, self.wcl.LED_INVERT, brightness)`
with
`self.strip = Adafruit_NeoPixel(self.wcl.LED_COUNT, self.wcl.LED_PIN, self.wcl.LED_FREQ_HZ, self.wcl.LED_DMA, self.wcl.LED_INVERT, brightness, 0, ws.WS2811_STRIP_GRB)`

within the file wordclock_tools/wordclock_display.py?

Please report, if that helps? ;)

Regards,
Bernd

protectionofficer, 2016-11-08:
Hi Bernd,

Yes, simply adding: 0, ws.WS2811_STRIP_GRB worked perfectly.
Can not thank you enough for you're vast knowledge and expertise.

Was it because I bought the wrong LED strip?

Darren

bk1285, 2016-11-08:
Ok cool. Thanks. Fixed in 98e6e2c

closed 2016-11-08 by bk1285