bk1285/rpi_wordclock #36: "am2303_rpi" no longer available in pip repository

oxivanisher, 2016-11-11:
It seems as the " am2303_rpi" is no longer available in the pip repository. I did not find another way to get the temperature and humidity sensor working.

bk1285, 2016-11-14:
How about this: https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/software-install-updated

Would be great to have short feedback, if and how it works...

Regards,
Bernd

oxivanisher, 2016-11-14:
I did all that already, but the rpi_wordclock still prints:
```
Imported plugin 5: "restart".
  Assumes no temperature sensor to be attached.
```
But, the tests from the Adafruit site are working.
```
root@wordclock:/home/pi/Adafruit_Python_DHT/examples# ./AdafruitDHT.py 2302 4
Temp=20.0*  Humidity=42.4%
```

bk1285, 2016-12-01:
Can you check with am2302 (instead of am2303)? To me, the docs seem to be a bit inconsistent here? http://rpi-wordclock.readthedocs.io/en/latest/doc_software_setup.html#optional-dependencies-to-readout-temperature-sensor

oxivanisher, 2016-12-23:
I was very busy the last weeks, but I will have a look at this the next week :+1:

bk1285, 2017-01-02:
Hey @oxivanisher 

Found the dependency to be wrong...
Fix is here: f4bbcfebc22b9185c9de7d1502e8837ff00c5580

Please reopen, if the issue persists.

closed 2017-01-02 by bk1285