bk1285/rpi_wordclock #37: NeoPixel dependency is too old to accept customized brightness values and correct RGB-settings

oxivanisher, 2016-11-11:
I did a clean install of the wordclock software as documented in the readme. With the current master of the https://github.com/jgarff/rpi_ws281x library, the wordlock is not working And with the mentioned commit https://github.com/jgarff/rpi_ws281x/tree/2f9e03c45ba0964029204db565ad9d6233b3a1a6, I get the following warning:
```
Exception AttributeError: "'Adafruit_NeoPixel' object has no attribute '_leds'" in <bound method Adafruit_NeoPixel.__del__ of <neopixel.Adafruit_NeoPixel object at 0xb633f370>> ignored
WARNING: Your NeoPixel dependency is to old to accept customized brightness values and correct RGB-settings.
```
Do you have another commit with these features available which you know that it is running?

bk1285, 2016-11-14:
Hi @oxivanisher,
do you have any further details, why the clock doesn't work with the current master from rpi_ws281x?
Regards,
Bernd

oxivanisher, 2016-11-15:
Yes, the error is the following:
```
root@wordclock:~# python /home/pi/rpi_wordclock/wordclock.py
Parsing /home/pi/rpi_wordclock/wordclock_config/wordclock_config.cfg
Setting up wordclock interface
Interface type set to gpio_low (False)
  Mapping button "left" to pin 17.
  Mapping button "return" to pin 22.
  Mapping button "right" to pin 24.
  Lock time of buttons is 0.2 seconds
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock.py", line 144, in <module>
    word_clock = wordclock()
  File "/home/pi/rpi_wordclock/wordclock.py", line 41, in __init__
    self.wcd = wcd.wordclock_display(self.config)
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_display.py", line 35, in __init__
    self.strip.begin()
  File "build/bdist.linux-armv6l/egg/neopixel.py", line 113, in begin
AttributeError: 'module' object has no attribute 'WS2811_SUCCESS'
```
I tried hacking around for some minutes, but since I have other problems with the project, I did not investigate it deeper at the moment. 😁

bk1285, 2016-11-18:
Just merged your pull request back. Great work, @oxivanisher I especially like the 3d printing part...!

bk1285, 2016-12-18:
Hi @oxivanisher, I recently setup another wordclock, but did not run into this issue.... Did you make you clock run meanwhile?

Regards,
Bernd

bk1285, 2016-12-21:
Please reooen, if the issue still exists.

closed 2016-12-21 by bk1285