bk1285/rpi_wordclock #38: Add the possible requirement of a "level shifter" to the documentation

oxivanisher, 2016-11-15:
I took me some testing and fiddling around to realize that I need a "level shifter" to drive my 5V LED's. They are flickering like this: https://www.youtube.com/watch?v=UHxVS8SkXOU

If possible, this should be added to the documentation somewhere. As recommended by the rpi_ws281x guys in this issue https://github.com/jgarff/rpi_ws281x/issues/127, I will use a 74HCT125.

bk1285, 2016-11-16:
Hi @oxivanisher thanks for reporting. I added your feedback to the troubleshooting section: https://github.com/bk1285/rpi_wordclock/commit/7c3a30268425ea77a963d2265a56b816ba1fb12c

Best,
Bernd

closed 2016-11-16 by bk1285

FocusInfinity, 2016-11-28:
FYI: This problem can be circumvented by installing older RPI noobs OS. I can recommend v1.9.2 published on May 31 in 2016. --> http://downloads.raspberrypi.org/NOOBS/images/

bk1285, 2016-11-28:
Great, @FocusInfinity! Thanks for the update! Did you also experience these difficulties?
Best, Bernd