bk1285/rpi_wordclock #40: Icons have wrong colors

oxivanisher, 2016-11-18:
https://github.com/bk1285/rpi_wordclock/commit/98e6e2c83bf7f94587b168a2b87d881e8cd92a64 worked like magic 👍 I now have the correct colors i.e. for the matrix plugin (it was red before, I think it was related to https://github.com/jgarff/rpi_ws281x/issues/122 ). 
BUT the icons still are in wrong colors, meaning red and green are again swapped. I am not sure if this problem is only because of my LED strip, or if its a general one. Am I the only one with this problem?

euchkatzl, 2016-11-18:
Try other values instead of ws.WS2811_STRIP_GRB ! 

Possible values are : 
WS2811_STRIP_RGB
WS2811_STRIP_RBG
WS2811_STRIP_GRB
WS2811_STRIP_GBR
WS2811_STRIP_BRG
WS2811_STRIP_BGR

oxivanisher, 2016-11-18:
No you probably misunderstand me. After the commit I mention, my colors are correct (since @bk1285 added ```ws.WS2811_STRIP_GRB``` to the strip initialization. But in the code to read the .png files, it is still as RGB and not GRB, which messes up the colors.

I think this should be configurable somehow anyways, since not anybody has the same LED strip. Something like
```
strip_color_ordering = "RGB"
```
in the configuration file.

The strip initialization in https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_tools/wordclock_display.py#L29 and methods like https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_tools/wordclock_display.py#L120-L122 then should use this configuration setting. So the color ordering will be configurable and not hard coded anymore.

bk1285, 2017-03-30:
Hi @oxivanisher,
can you check, if an update of the third-party-dependency fixes your issue?:
https://github.com/jgarff/rpi_ws281x

Best,
Bernd

closed 2018-01-04 by bk1285

oxivanisher, 2018-01-04:
I am sorry to have forgotten to check this. Yes, it fixed my clock AFAIK. Since i have some dead LEDs, I am still waiting for replacements and can not do a "live" check, but I believe the closing of the issue fine.

bk1285, 2018-01-04:
Ok, cool. Please reopen in case of recurrence...