bk1285/rpi_wordclock #44: Raspberry Pi Zero

Nthl25, 2016-11-29:
Hallo,

kann mir jemand von euch sagen, ob die Wordclock auch mit einem Raspberry Pi Zero funktioniert?
Leider gibt's häufig nur noch den Pi 3 zu kaufen, der ist aber in meinen Augen völlig überdimensioniert.

Grüße
nthl25

bk1285, 2016-11-30:
Bin kurz davor das zu testen... Werde mich mit Updates melden.

Regarding the RPi3: Agreed, this one is overdimensioned. RPi0 would be great... ;)

HeerNMeester, 2016-11-30:
I'm building a version based on the RPi0. A Dutch 11x10 to be precise. So far I have installed the software, which gave no problem (except the AM2302 temperature sensor, but that a known issue). I don't expect further problems. My time is limited, so progress is slow.

Nthl25, 2016-12-01:
Hallo zusammen,

ich bekomme beim installieren von python-scipy auf meinem Pi0 folgenden Fehler:

Es wurden 25,6 MB in 2 min 55 s geholt (145 kB/s).
E: Fehlschlag beim Holen von http://mirrordirector.raspbian.org/raspbian/pool/main/p/pillow/python-imaging_2.6.1-2+deb8u2_all.deb  404  Not Found [IP: 5.153.225.207 80].

Habt ihr eine Idee, was ich da machen kann?

Grüße

bk1285, 2016-12-01:
@Nthl25 , Moved your issue here https://github.com/bk1285/rpi_wordclock/issues/45 since its not related to the Pi0

Nthl25, 2016-12-02:
@bk1285 , okay. First I thought it was a RPi0-Problem. So I postet it in this issue. Hope anyone can help :)

HeerNMeester, 2016-12-11:
Yesterday I started my word clock with RPi0 for the first time. It it still a bread board design, but it worked fine (Many thanx to @bk1285 !!). 
Only a few issues which can be resolved, but are not related to the use of the RPi0. (I also use a cheap China LED strip, so my colors are inverted as well (need to update the software) and my clock seems to be running one hour behind (daylight saving time??).)
Temperature and light sensor need to be implemented still. I do not use a level shifter at the moment.

bk1285, 2016-12-12:
Hi @HeerNMeester thanks for the feedback! I had the software running on a RPi0 as well (without attaching the actual LEDs). So I close this issue and add a note to the docs concerning the compatibility with the RPi0.

closed 2016-12-12 by bk1285