bk1285/rpi_wordclock #45: Error during install of 3rd party dependency

bk1285, 2016-12-01:
Reporter by @nthl23

Hallo zusammen,

ich bekomme beim installieren von python-scipy auf meinem Pi0 folgenden Fehler:

Es wurden 25,6 MB in 2 min 55 s geholt (145 kB/s).
E: Fehlschlag beim Holen von http://mirrordirector.raspbian.org/raspbian/pool/main/p/pillow/python-imaging_2.6.1-2+deb8u2_all.deb 404 Not Found [IP: 5.153.225.207 80].
Es existiert jedoch diese Datei mit einer anderen Kennung (?) (deb8u3_all.deb anstelle von deb8u2_all.deb).

Habt ihr eine Idee, was ich da machen kann, um die zu installieren?

Edit: Dieses Problem habe ich sowohl bei meinem Pi0 als auch bei meinem Pi2.

Grüße nthl23

Nthl25, 2016-12-02:
It's really easy to solve...

> sudo apt-get update
helps

bk1285, 2016-12-02:
@Nthl25 Great! Thanks for resolving... :)

closed 2016-12-02 by bk1285