bk1285/rpi_wordclock #46: type = gpio_high but it still says "Pin 17 pressed"

Nthl25, 2016-12-03:
Hi,

I set the "type" in my configfile to gpio_high but when starting the wordclock-script, “Pin 17 pressed” is logged all the time. Do you have any ideas what this could be?

nthl

bk1285, 2016-12-03:
you tested with gpio_low as well, right?

Nthl25, 2016-12-04:
I tried both. But I always get the message "Pin 17 pressed".

bk1285, 2016-12-04:
:/ ok. I'll try to reproduce with the latest version of the software as soon as I can...

Nthl25, 2016-12-04:
I get the same problem if  i  connect my 'new' Pi to my 'old' wordclock. Do you know how i  can  install an old  version of your wordclock?

Nthl25, 2016-12-05:
It is not an error in your wordclock software, sorry for my mistake.

I think it was something lika a cold soldering point on my LED stripes.
Now It is working with your latest software!

closed 2016-12-05 by Nthl25

bk1285, 2016-12-05:
Ok. Thanks for resolving!