bk1285/rpi_wordclock #55: type = gpio_high but it still says "Pin 17 pressed"

Tihmann, 2016-12-19:
I did a fresh install of your software. I have just connected the LED strip, no buttons. Parameter was set to gpio_high but I still get "Pin 17 pressed".....There is no soldering on the LED strip yet

Yoghurt02, 2016-12-19:
Same problem here. 
Couldn't figure out what to change until now... 
Help would be much appreciated.

bk1285, 2016-12-19:
No good.... Could you checkout a previous version, if it works? The git command

    git checkout HEAD~3

should work out ... Replacing 3 with the number of your choice...

Yoghurt02, 2016-12-19:
Thanks for the fast reply!
I'm very new to git so I don't exactly understand in which folder I have to be or what commands come first. 
I've only entered "git checkout HEAD~2" 
Would you please describe it for a beginner?

euchkatzl, 2016-12-20:
@bk1285 
I think the problem are my virtual buttons used for the remote control. We should try to add an opportunity to disable that.

euchkatzl, 2016-12-20:
Hey,
i got things working with a fresh install. There were several problems :

1. We have to correctly initialize GPIO with the correct state
2. Remote Control did cause the problem above when gpio_high was set

When you have no real buttons you have to set type = gpio_high !

Look at the following commit 
b687aa1dadd7c0bb448f5da47dfef36cf9b7a58d

@bk1285 : Can you merge it back ?

bk1285, 2016-12-20:
Hey! Could you please check the latest commit https://github.com/bk1285/rpi_wordclock/commit/73c285597c1241d7ca11b6e3ee1428edf71b926a

.... leaving the virtual interface enabled, since I like it a lot! ;) 
... but many thanks @euchkatzl for helping out!!!

Please report back, if the fix works, since I can only partially verify with my own hardware-configuration.

Yoghurt02, 2016-12-21:
Hey guys,

thanks a lot for your effort.
I will be able to check it tonight and I will let you know if it works!

Yoghurt02, 2016-12-21:
Hey,
I've replaced the wordclock_interface.py with the one from 73c2855
Type is set to GPIO_High but it's still acting like Pin 17 is pressed.
Did I miss something?

Yoghurt02, 2016-12-21:
Alright, now that I did all the changes from @euchkatzl it works!
Thank you so much guys!
I'm super happy.

Tihmann, 2016-12-22:
Tried the latest commit 73c2855. Still says Pin 17 pressed. What further changes do I need to do ?@bk1285 Could you merge back the changes from euchkatzl ?

euchkatzl, 2016-12-22:
The Problem is that you have to initialize the pins with the correct values.

Look at commit  a8cfde63a1f9593c2d8385f14d314036e5357932

`
GPIO.setup(self.button_left, GPIO.OUT)
GPIO.setup(self.button_return, GPIO.OUT)
GPIO.setup(self.button_right, GPIO.OUT)        

GPIO.output(self.button_left, not self.alter_interface_state) 
GPIO.output(self.button_return, not self.alter_interface_state)
GPIO.output(self.button_right, not self.alter_interface_state)

GPIO.setup(self.button_left, GPIO.IN)
GPIO.setup(self.button_return, GPIO.IN)
GPIO.setup(self.button_right, GPIO.IN)
`

That are the important lines !

Tihmann, 2016-12-22:
What are the other commits

b687aa1
0707f52
a8cfde6

for ? Do I need them as well ? I would like to use the clock without any buttons

euchkatzl, 2016-12-23:
The other commits are just for config changes so that you have not my bavarian config.

You can simply checkout my fork in another directory and then link Fontdemo.py again. Should work then.

Or you manually add the code lines above in wordclock_tools/wordclock_interface.py . 

Or you can friendly ask @bk1285 if he can include this fix too 

@bk1285 i don't know if its the right way to do this but it helps fixing that issue. The Problem is that one of these pins have a logical 1 as value. So we first have to initialize it as OUT, then set the value to the opposite of our used config (for gpio_high we set it to 0). And afterwards we initialize it back ti an IN Pin. Perhaps the better solution is to make it configurable to either use buttons or not (and then not to initialize the GPIO's as IN )

Tihmann, 2016-12-27:
@euchkatzl Just as a follow up. I checked out your fork. With that fork the wordclock is working perfectly. No more PIN pressed errors. @bk1285 Could you merge euchkatzl´s modifications into your git

bk1285, 2016-12-31:
@euchkatzl @Tihmann @Yoghurt02 

Hi all!

thanks for the input and feedback! So if I got it right, it's about adding a8cfde6?
If you confirm, I'll merge and them onto the master branch. For new and experimental stuff, I'll add a development branch. New things can be tested there, before they go to the master branch. ;)

Regards,
Bernd

bk1285, 2016-12-31:
One of the first things to be tested on the dev-branch could be  a fix, allowing to enable the virtual buttons for gpio_low and (!) gpio_high :)

Tihmann, 2017-01-02:
@bk1285 Hi, I think the main issue is a8cfde6. However I did a fresh install from @euchkatzl 's fork. That fork is working without any problems

closed 2017-06-19 by bk1285