bk1285/rpi_wordclock #56: Button functions

Yoghurt02, 2016-12-23:
Hey all,
I've now also added three buttons.
I've figured out that the left button changes the clock colors.
But what do the other buttons do?
How can I enter the other plugins like Matrix, temperature and so on?

Yoghurt02, 2016-12-23:
I've seen that Pin 22 should change the plugin but after I've pressed it it always writes "Running plugin time_default".
In the config all the plugins are set true and in the startup it lists the plugins from 0 to 10.
Anybody an idea what is missing?

bk1285, 2016-12-25:
Hm... Expect the wordclock to behave as outlined within this video: https://youtu.be/wcLQDykRBbM 
Anything else would be wrong and strange... :/

Can you activate the color wheel by pressing the 3rd button?

Regards,
Bernd

Yoghurt02, 2016-12-27:
Hi Bernd, 
3rd Button is the "return" button, right?
It doesn't change the plugin. Sometimes the lights are flashing. If I press again, they stop.
If i check it in the command window, for each "return"-Button press it says "Running plugin time_default". 
For me that means button press is recognized but it's not changing the plugin.

bk1285, 2017-01-03:
Do you have besides the default plugin other plugins enabled? If so, you should be able to change between them.

Regards.

Yoghurt02, 2017-01-03:
Yes. In the config file I have enabled all of them...

bk1285, 2017-06-19:
Please reopen, if necessary.

Best,
Bernd

closed 2017-06-19 by bk1285