bk1285/rpi_wordclock #61: Pin 17 always pressed

danielschmaltz, 2017-03-03:
I do not want to use Buttons, but when I set  type to gpio_high it does not make a difference.

Easiest Setup: Pi B+, no Buttons, Edimax Wifi and latest Jessie

It seems that the led stripe works fine, cause it is obviously going through all plugins...but all the time...

Any idea?
Thanks

bk1285, 2017-03-04:
Hey @danielschmaltz 

can you give it a try with the latest develop branch of the repository, reporting back, whether this works out?

`git checkout develop`

Regards,
Bernd

danielschmaltz, 2017-03-05:
Thanks Bernd, I tried the develop branch, but nothing changed. Now I  installed @euchkatzl 's fork and it works lika a charme. I guess theres a problem with the GPIO states.

Thanks for this repo.
Best, Daniel

closed 2017-03-05 by danielschmaltz