bk1285/rpi_wordclock #63: Plugin Sunset failed Import

mblonski2015, 2017-03-25:
Hey,
I gave away my World Clock as a gift, but now I got the feedback, that the sunset is missing.
After I SSH'ed me to the pi and watched the startup, I do got the error, that sunset failed to import.
I downloaded the plugin again. renamed the folder and uploaded to the pi, but even the "new" plug in failed to be imported.
Is there any log or anything I should check? Honestly I have no clue where to start my search.
Any help is highly appreciated.

Many thanks in advance
Mathias

mrksngl, 2017-03-28:
A good starting point is telling us at least something about the error, i.e. all the output of the python process.

mblonski2015, 2017-03-28:
Hello,

thanks fort he feedback.
At startup is just says

„import Sunset failed“

I’m sorry, but I’ve produced it as a gift and as I handed it over I found out that the sunset is missing and unfortunately I don’t have access to it right now.

Since I re downloaded the plug-in and renamed the folder to sunset 2 and it comes up with the same error, I assume, that the plug in is fine. Do you know what is checked during the import?
Mathias

euchkatzl, 2017-05-05:
Did you install all dependencies ?
I think for the sunset Plugin you need :
`pip install astral`

When it is nor working either do the following:
Import traceback module 
`import traceback`
add this in the in line 77 (before printing "Failed to import plugin") in wordclock.py :
`traceback.print_exc()`

Then you will get a more detailed error.

bk1285, 2017-06-19:
Hi mblonski2015

please reopen, in case you still experience these issues and you can provide new information as proposed by @mrksngl and @euchkatzl 

Best,
Bernd

closed 2017-06-19 by bk1285