bk1285/rpi_wordclock #64: Exact Frame Size?

adamenatsky, 2017-04-10:
Hi. I'm using a raspbery pi zero w and so far the software is running great but i'm having trouble finding out the exact measurements used for the LED framing. In the config file says the absolute size is listed at 450mm x 450mm with a 20mm margin from the edge of the stencil to the minute LEDs. I traced that out on some paper and noticed that those dimensions don't quite match up to what is shown in the build pictures you have posted. For example the margin in the photos is large enough to box in the Raspberry Pi. Should I just assume outside of the 450x450mm is purely a builder's choice as to the size? Thanks for your help. Looking forward to completing this project!

bk1285, 2017-04-10:
Hi @adamenatsky,
the margin relates to the minute-LEDs. The stancil is 450x450mm. The actual display size is 300x300mm
All other measures (spacing between the letters) relate to that.

Regards,
Bernd

adamenatsky, 2017-04-10:
@bk1285 Cool thanks for the clarification. I'll take another try the layout.

closed 2017-04-10 by adamenatsky