bk1285/rpi_wordclock #65: rasperry pi zero not compatible ?

Carlos464, 2017-06-05:
Hello,

I am trying to get the wordclock up and running with an raspberry pi zero, but that does not work. The software is running, but none of the LEDs is getting on.

The same config with an rpi2 is running fine.

Do I have to tweak the software?
Anyone else having these issues?

Thanks in advance.

Carlos

adamenatsky, 2017-06-10:
Hi Carlos. I have gotten the clock to start working on a pi zero w. A few possible bugs I noticed after tinkering are that I had to disable the opening message in the config file for the clock to start working, not sure why. Second issue I ran into was that the clock would only work when started from the cron at boot. When starting it manually from the command line the lights would just blink randomly and show me a jumble of colors. Otherwise it works fantastically on the pi zero. Let me know if I can help with any questions you have. I'm not an expert just a hobbyist by the way.

Carlos464, 2017-06-12:
Hi adamenatsky,

thanks for your reply.

Do you use the same GPIO Pins as mentioned in the docs?
The opening message is already deactivated and the program starts via cron at boot.

best regards
Carlos

adamenatsky, 2017-06-12:
Yes I followed Bernd's wiring and GPIO pin usage. My python skills are quite novice so I didn't want to mess around with the source code until I got comfortable with the device working in its normal state.

bk1285, 2017-06-14:
Hi all, thanks for your input!

Find first reports and discussions on Pi0 here:
https://github.com/bk1285/rpi_wordclock/issues/44
Hope it helps you make the clock run.

Just in case you find out futher tricks, it's great, if you share them here.

Best,
Bernd

bk1285, 2017-06-14:
Did you take this page into consideration: https://github.com/bk1285/rpi_wordclock/blob/master/docs/doc_troubleshooting.rst

... a disabled soundcard for only one of the two Pis might be an issue?

adamenatsky, 2017-06-14:
Hey Bernd I'm new this whole github community. Should I add any notes I posted here to he #44? Or is the link you put up good enough?

And yes I did look at the troubleshooting page early on. Only issue I ran into listed there was the GPIO Pin 17 thing. But that was cleared up quickly once I connected the buttons and LEDs. 

Thanks for your all you work on this project! Its been fun getting it up and running.

Carlos464, 2017-06-14:
Hi Bernd.
I just tried the suggestions from your link.
Still no luck.

Anyone any ideas?

best regards
Carlos

Carlos464, 2017-06-16:
@adamenatsky

Can you do me a favor and post the output of 

`gpio readall`

on your working raspberry pi zero.

Thanks in advance.

best regards
Carlos

adamenatsky, 2017-06-19:
@Carlos464 I will do so hopefully later this evening or tomorrow. I'm just now resoldering my buttons  in a permanant placement and will have the pi up and running then.

Carlos464, 2017-07-01:
Hi all,

it's still not working with the rPi0.
Would love to get it up and running.

@adamenatsky
Any chance to upload a screen of the output of

`gpio readall`

best regards
Carlos

adamenatsky, 2017-07-05:
@Carlos464 Apologies for the delay. As I'm not good at soldering, I damaged some of the pixel connections while making permanent wiring placements and I am in the process of redoing them. I'll get your readall as soon as I can.

Carlos464, 2017-07-10:
@adamenatsky 

I started over from scratch and now it is working with the raspberry pi zero.
From smoke to smother. Now the rPi hangs randomly.
I will open a new thread.

best regards
Carlos

ruedli, 2017-12-27:
I installed jessi-lite (2017-11-29-raspbian-stretch-lite) on a "raspberry pi zero w" and found the clock would not compile, complaining about missing modules: freetype and monotic.

It showed:
Traceback (most recent call last):
  File "wordclock.py", line 8, in <module>
    import wordclock_tools.wordclock_display as wcd
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_display.py", line 1, in <module>
    import fontdemo
  File "/home/pi/rpi_wordclock/wordclock_tools/fontdemo.py", line 30, in <module>
    import freetype

and

Traceback (most recent call last):
  File "wordclock.py", line 9, in <module>
    import wordclock_tools.wordclock_socket as wcs
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_socket.py", line 5, in <module>
    from wordclock_interfaces.event_handler import event_handler as eh
  File "/home/pi/rpi_wordclock/wordclock_interfaces/event_handler.py", line 3, in <module>
    from monotonic import monotonic as _time
ImportError: No module named monotonic

The resolution for me was (sorry, if this should have be done easier, I am new to raspberry and python):

cd ~
git clone https://github.com/rougier/freetype-py.git
git clone https://github.com/atdt/monotonic.git

cd ~\freetype-py
sudo python setup.py build
sudo python setup.py install

cd ~\monotonic
sudo python setup.py build
sudo python setup.py install

after installing you can remove the cloned monotonic and freetype installation directories.

Also, an installation step to set your current timezone and DST was missing, which can be done using sudo raspi-config in the localization option.

closed 2018-01-04 by bk1285