bk1285/rpi_wordclock #66: Config file help

standi01, 2017-06-18:
Hi Bernd

Thanks for this clock, I have been looking at ways of making one for quite a while now then I found your project.

I have managed to make this clock although I used a 60 led per metre strip to create a smaller clock. I have set up the clock to run in English although some of the plugins are not working correctly. 

I am having trouble with the config file. I will be honest and say I don't know very much about python and programming in general. 
I have managed to copy the example config file and it's in the same directory but when I run the software it says no linked config found, using example.cfg instead.
If I remove the example config file the software can't find any and won't run. 
What do I need to do to link it?
I have had to edit the example config file to get it to run.

Also any tips, links or guides that I could use to set the config up or even how to create/ add new plugins would be appreciated.

Thanks for any help that anyone is willing to give.

Mark

stefancarlton, 2017-06-19:
have you checked permissions and ownership on the new file? Did you name it wordclock_config.cfg and keep it in the wordclock_config folder (https://github.com/bk1285/rpi_wordclock/blob/master/wordclock.py#L24)?

I created an English language version a while back, but simplified the config for my needs: https://github.com/stefancarlton/rpi_wordclock/blob/master/wordclock_config/  (the simplification came in time_default where I simply hard-coded the ranges: https://github.com/stefancarlton/rpi_wordclock/blob/master/wordclock_plugins/time_default/time_english.py). I also tweaked the hardware buttons by removing the constant polling to lower the CPU usage.

standi01, 2017-06-19:
Hi Stefan Carlton

I have checked the config file. It's as it should be. All I did was copy the example config and removed example from the name. When I run the software it can't see it and uses the example one.

What do you mean by permissions and how do I check?

bk1285, 2017-06-19:
Hey @standi01

here we go.... https://www.linux.com/learn/understanding-linux-file-permissions

Your config file should be named wordclock_config.cfg and should be readeable by root.
... to set the permission try to enter
chmod 777 wordclock_config.cfg

Please report back, if this resolved your issues...

Best,
Bernd

standi01, 2017-06-19:
Bernd

Thanks for the information​, I had a look at the file and have now managed to resolve the config issue.

Another quick question if I am allowed, I have  copied and edited the plugin for time in words German and changed the relevant values to English. I placed the folder in the plugins folder. How do I then install onto the clock to test.
I have changed the configuration file to read time_in_words_english but it can't find it and reverts to German.
That's once again

Mark

standi01, 2017-06-20:
Basically my current state is I currently have the clock set up as Bernds files.
 I have changed the RSS feed and it displays English text. I have set the location up for the weather app with no main issues.
I can't get the sunrise plugin to display the times in English (tried adjusting the config to look in the english.py file of the default time folder. Fails to load plugin or reverts back to the German text.
I also can't get the matrix to display the time in English.
I have tried to get the time in words into English by amending the German config files and checking the values to English.
Failing miserably.

Advice on the plugins would be appreciated as I would like to create one (clock face and a digital time display (no reason but would will be a project)) but need to learn my way around the current ones first.

Thanks all

standi01, 2017-06-21:
Running from my repository with my plugin config I get

Parsing /home/pi/rpi_wordclock/wordclock_config/wordclock_config.cfg
Setting up wordclock interface
/home/pi/rpi_wordclock/wordclock_tools/wordclock_interface.py:105: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.virtual_button_left, GPIO.OUT)
/home/pi/rpi_wordclock/wordclock_tools/wordclock_interface.py:106: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.virtual_button_return, GPIO.OUT)
/home/pi/rpi_wordclock/wordclock_tools/wordclock_interface.py:107: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.virtual_button_right, GPIO.OUT)
Interface type set to gpio_low (False)
  Mapping button "left" to pin 17.
  Mapping button "return" to pin 22.
  Mapping button "right" to pin 24.
  Mapping button "virtual_left" to pin 7.
  Mapping button "virtual_return" to pin 8.
  Mapping button "virtual_right" to pin 11.
  Lock time of buttons is 0.2 seconds
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
  Assumes no temperature sensor to be attached.
Imported plugin 0: "weather_forecast".
  INFO: No activate-flag set for plugin rainbow within the config-file. Will be imported.
Imported plugin 1: "rainbow".
Imported plugin 2: "time_as_words_dutch".
Failed to import plugin time_matrix!
Imported plugin 3: "shutdown".
Imported plugin 4: "ip_address".
Imported plugin 5: "restart".
Imported plugin 6: "matrix".
  INFO: No activate-flag set for plugin time_as_words_english within the config-file. Will be imported.
Failed to import plugin time_as_words_english!
Imported plugin 7: "time_as_words_german".
  Selected "time_default" as default plugindont
Imported plugin 8: "time_default".
Imported plugin 9: "feed_parser".
Failed to import plugin sunrise!
Imported plugin 10: "tetris".
Running plugin time_default.

I am assuming the error is caused by the no active flag but I don't know where to rectify this.

standi01, 2017-06-23:
Found some errors in the code but still not 100% 
When I run my version I get the following

pi@raspberrypi:~/rpi_wordclock1 $ sudo python wordclock.py
Parsing /home/pi/rpi_wordclock1/wordclock_config/wordclock_config.cfg
Setting up wordclock interface
/home/pi/rpi_wordclock1/wordclock_tools/wordclock_interface.py:105: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.virtual_button_left, GPIO.OUT)
/home/pi/rpi_wordclock1/wordclock_tools/wordclock_interface.py:106: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.virtual_button_return, GPIO.OUT)
/home/pi/rpi_wordclock1/wordclock_tools/wordclock_interface.py:107: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.virtual_button_right, GPIO.OUT)
Interface type set to gpio_low (False)
  Mapping button "left" to pin 17.
  Mapping button "return" to pin 22.
  Mapping button "right" to pin 24.
  Mapping button "virtual_left" to pin 7.
  Mapping button "virtual_return" to pin 8.
  Mapping button "virtual_right" to pin 11.
  Lock time of buttons is 0.2 seconds
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
  Assumes no temperature sensor to be attached.
Imported plugin 0: "weather_forecast".
  INFO: No activate-flag set for plugin rainbow within the config-file. Will be imported.
Imported plugin 1: "rainbow".
Imported plugin 2: "time_as_words_dutch".
Imported plugin 3: "time_matrix".
Imported plugin 4: "shutdown".
Imported plugin 5: "ip_address".
Imported plugin 6: "restart".
Imported plugin 7: "matrix".
  INFO: No activate-flag set for plugin time_as_words_english within the config-file. Will be imported.
Failed to import plugin time_as_words_english!
Imported plugin 8: "time_as_words_german".
  Selected "time_default" as default plugin
Imported plugin 9: "time_default".
Imported plugin 10: "feed_parser".
Imported plugin 11: "sunrise".
Imported plugin 12: "tetris".
Running plugin time_default.

I am also getting some other random errors (no moon in sunrise, error in Tetris)

closed 2017-06-24 by standi01

standi01, 2017-06-24:
Sorted it out. Sunrise, time matrix is now in English and I have added time in words English. New to Github so I don't know how to link/merge etc with the main project.
Now time to try to create my own plugins

bk1285, 2017-06-24:
Great! :)