bk1285/rpi_wordclock #69: Farbwechsel

coolio210, 2017-08-29:
Hallo, 
ich kann leider nicht programmieren. Ich habe mir eine Wortuhr ohne Buttons nachgebaut. Diese funktioniert auch. Ich möchte aber nun, dass die LED-Farbe regelmäßig wechselt. Kann mir jemand sagen, in welcher .py-Datei ich was ändern muss, damit das funktioniert? 
Danke im Voraus.

euchkatzl, 2017-08-29:
Wie oft soll das denn wechseln ? und in welche Farben soll es wechseln ?

coolio210, 2017-08-29:
Am liebsten einmal durchs Regenbogenspektrum so langsam überfaden. 
aber wenn das nicht geht, würden mir auch Farben wie rot, gelb, grün, orange, violett und blau reichen, die durchwechseln.

euchkatzl, 2017-08-30:
Den Effekt den du haben willst gibt es ja eigentlich schon (Drücken des Rechten Buttons)
Da du aber keine Buttons hast musst du eigentlich nur den Klick manuell auslösen.

Dazu änderst du folgende Datei (Die Änderungen sind im angehängten Zip):
~/wordclock_plugins/time_default/plugin.py

[plugin.zip](https://github.com/bk1285/rpi_wordclock/files/1262503/plugin.zip)

euchkatzl, 2017-08-30:
Ich würde aber an deiner Stelle auch mal versuchen die Remote Steuerung zum Laufen zu bekommen.

Dann kannst du den Effekt per Handy ein und ausschalten und auch die Uhr im vollen Umfang benutzen.

coolio210, 2017-08-30:
hallo euchkatzl,
vielen Dank für Deine schnelle Reaktion und die Datei. Werde ich gleich mal ausprobieren. 
Daß es auch eine Remote-Steuerung gibt, war mir gar nicht bekannt! Ist anscheinend völlig an mir vorbei gegangen. Hatte nur irgendwo gelesen, dass es mal als Erweiterung/Verbesserung vorgeschlagen wurde. Wusste nicht, dass es schon umgesetzt ist. 
Das wäre ja noch viel geiler, wenn das funzen würde. Wo kann ich mich denn da einlesen?

coolio210, 2017-08-31:
Die Datei funktioniert leider nicht. Nach der Begrüßung "Hi all" läuft der Schriftzug "Error in time default" durch.

euchkatzl, 2017-09-01:
Welchen Branch benutzt du denn ? Den Develop oder den Master ?

Blöde Frage aber du hast die Datei schon entpackt und die py Datei hinkopiert oder ?

Für Remote Control gibts es zwei Möglichkeiten :

1. Wenn du den Master Branch benutzt : https://github.com/euchkatzl/RaspberryWordclock-RemoteControl
2. Wenn du den Develop Branch benutzt kannst du mal meinen Fork probieren : 

- https://github.com/euchkatzl/rpi_wordclock/tree/develop
- Da ist die Steuerung unter {{IP}}:8081 erreichbar

Ich weis nicht ob @bk1285 schon eine neue Lösung für die Web Oberfläche hat ?

coolio210, 2017-09-01:
Hi, 
ich habe die Version die ich vor ca. 1/4 Jahr von der Github Seite bei bk1285 heruntergeladen hatte => also Master Branch, oder?) . Eine Weboberfläche hat diese Version nicht. 

Ja, ich habe die entpackte .py-Datei so wie von Dir oben beschrieben in das Plugin-Unterverzeichnis kopiert (die Originaldatei habe ich in einem anderen Unterverzeichnis gesichert). 

Ich muss die Tage eh nochmal ran. Ich probier mal ein bisschen rum. Danke für die Links! Ich bleibe dran und werde berichten.

coolio210, 2017-09-07:
So, 
ich habe jetzt noch mal alles neu gemacht (auch das Hardware-Wiring), eine komplette Neuinstallation des Betriebssystems auf dem Raspberry Pi 3 und habe die o.g. Datei durch Deine ersetzt. 

ES LÄUFT !!!

Super!!! 
Werde mich jetzt mal der Handysteuerungsvariante zuwenden. Vielen Dank für die Unterstützung !!!

bk1285, 2017-10-21:
Ok, cool! Klingt gut :) Ich schließe dieses Issue dann jetzt einfach mal.

VG,
Bernd

closed 2017-10-21 by bk1285