bk1285/rpi_wordclock #72: Power supply

DavaD85, 2017-10-01:
Hi
I'm planning to buy the power supply for led strip and raspberry pi III, do you have idea about sizing? Di you have an example of power supply I need? Thanks

bk1285, 2017-10-01:
Hi @DavaD85,

here we go:
http://astore.amazon.de/araspbepibase-21/detail/B00KETLBAU

http://astore.amazon.de/araspbepibase-21

Best regards,
Bernd

DavaD85, 2017-10-01:
Great!! Thanks a lot I'll buy it and start soldering everything.

DavaD85, 2017-10-01:
Docooler AC 100V ~ 240V a DC 5V 12A 60W Tensione Trasformatore Interruttore Alimentazione per Striscia LED https://www.amazon.it/dp/B01M0VFYUT/ref=cm_sw_r_cp_apa_mZq0zbFK46MB7

Can be this ok? I've the power not near the clock and with this I can attach a long cable, 60w 12a 5v

bk1285, 2017-10-01:
Should be fine... :) Already with 10A, you can continuously run all 114 LEDs at full brightness. To show the current time, you need around 20-25 of them. Only some plugins require more LEDs (Tetris, etc...)

Best,
Bernd

DavaD85, 2017-10-01:
Ok so finally I bought this

https://www.amazon.de/gp/aw/d/B00MWQF6M2/ref=mp_s_a_1_1?__mk_de_DE=ÅMÅZÕÑ&qid=1506874273&sr=8-1&pi=AC_SX236_SY340_FMwebp_QL65&keywords=lpv+100+5&dpPl=1&dpID=3167vJ9lKqL&ref=plSrch

I've to put it in a small box in the wall, I think this is perfect :)

ruedli, 2017-10-07:
I did not like the rather high current of the 5V supply.... So considering another approach: I will use "an old" 23V notebook adapter and use 5 stepdown convertors for the matrix and one for the PI. 

So: then (if I use 90% efficiency, they even claim 97.5% but I do not believe that) the max current for all leds on in the matrix from the notebook power supply is **less than 2A.** I will try and use these: 

https://www.aliexpress.com/item/Mini-DC-DC-12-24V-To-5V-3A-Step-Down-Power-Supply-Module-Buck-Converter-Adjustable/32767397427.html

It should be much better in terms of voltage loss over the supply lines. If you hook up a 5V PSU directly, due to the large current, the voltage at the strips will be significantly less than 5V. With the step down convertors right next to the strips this is no longer the issue AND the step down convertors will happily take any voltage input as long as it is higher than 6V and convert it to exactly 5V. This will never be an issue with the 23V input and the lower current.

So far the theory.... I've only used this with Arduinos and around 500 animated LEDS in a matrix (not pixels, I used 4 HT16K32 drivers for 128 LEDS each) in my pinball machine. That worked great. I will keep you posted. If someone thinks this is a **bad idea for** the clock for things I have overlooked: **please comment.**

bk1285, 2017-10-21:
Hi @ruedli,
meanwhile, a huge amount of people built the raspberry-based wordclock, using the power supply, as proposed in the documentation of this project. Since I'm not an expert in electronic, I cannot comment on your plans in any way.

Regards,
Bernd

ruedli, 2017-10-21:
Hi Bernd,

Thanks for looking into it, I can understand your point. For that matter: I
am an electronic engineer, but the clock doesn't know that ;-)
I'll give it a try and see if it flies.

Regards,

Ruud

ruedli, 2017-12-27:
So, the clock is running for two days on the mentioned small step-down convertors (less then 50 cents each).

I am now using 4: one for the pi and three for the strips. The strip ones serve each a separate part of the matrix:

- convertor 1 left 3x10 LEDS + the left minutes LED, so 32 LEDS in total.
- convertor 2 middle 4x10 LEDS, so 40 LEDS in total. 
- convertor 3 right 3x10 LEDS + the right minutes LED, so 32 LEDS in total.

Each convertor delivers 5V and 3A max, can be supplied by anything DC from 7V to 29V (I am using an old DELL notebook adapter providing 19.5V). You need the same mount of Watts as specified for the clock, but now the needed current will be lower because of the higher voltage.

https://www.aliexpress.com/item/5V-3A-Mini-Step-Down-Power-Supply-Module-DC-DC-Converter-Input-7V-28V-Output/32825744436.html

In this way the voltage suppllied at the strip and RPI is indeed 5V, less loss over the line, wince the current provided by the PSU is lower and is regulated to be precisely 5V (in fact mine are between 4.96V and 5.10V).

I will add small cooling sinks, but even when all the LEDS are "full whites", for many hours, these step down convertors get "warm", but not too much. At the normal clock display mode, they do not even get warm.

I will add two additional 6V drivers for my 12 servo's... 

As Bernd mentioned, I provide the information as-is, without guarantees for success, however, from an electrical engineering perspective this solution made sense to me, which is why I provide this feedback.

closed 2018-01-04 by bk1285