bk1285/rpi_wordclock #73: Clock is stable for a while, but after a week or so....

Arjan444, 2017-10-20:
Dear All, I have a strange phenomenon. My Clock is completely build as described in this repository including level shifter, lots of extra VCC and GND wires to make sure there is enough grounding and voltage distributed over all LED stripes, 5V/10 Amp power source, resistor before the data hits the first led. Everything works as a charm, but then after a week or so I see strange colours appear and random LEDs turn on. Switching the power off/on helps for a while but after a day or so the same problems reoccurs.

Is it the Pi? Is it the level shifter? Could it be the Leds? I'm puzzled. Looks to me like a fuzzy problem (everything works for multiple days and then of a sudden I see strange led behaviour). 

Please help with a sensible, logical explanation. 

Thanks! Arjan

bk1285, 2017-10-21:
Hi @Arjan444,
indeed, that's a strange thing. You're the first to report this kind of issue and I suspect the hardware causing this behavior. I currently don't have any idea, what could be the reason for this... 
But feel free to upload some pictures of your wiring here, to get the discussion started.

Regards,
Bernd

mrksngl, 2017-10-21:
I've seen similar things in another project, and it turned out to be a problem of the clock (but in that project I have a really fast update, i.e. clock speeds at the very upper limit of the strip) and especially the level shifter (one of them was too slow, the other one too weak). It should also be noted, that the strip I used had seperate clock and data wires, which can also introduce some more trouble.

I would recommend to first check your soldering points again. And you should make sure there is litte cross talk on the wire (i.e. do not make the data/clock wire longer than needed -- especially: do not loop it).

Which level shifter did you use? I don't really get "resistor before first led", can you tell me more about you wiring? Or could you provide a link to the document/issue describing your setup?

It could also be an issue of your power supply which could either be too weak or at least too unstable. You could try to add a capacitor (e.g. 4.7uF) right before the led strip begins.

Is your problem affecting all LEDs or only some of later points in the LED chain, i.e. assuming you used bernd's wiring, is the error occuring at all rows or some later rows?

Regards,
mrks

Arjan444, 2017-10-21:
Thank you mrks. The strip I use, uses separate clock and data wires. I think you confirm my suspicion that it had something to with the wiring. I use clips to connect the strips, and since they are  15 cm I have to curl/loop them a bit. And that may influence the clock signal. 

For a level shifter I used the 74AHCT125 . With 'resistor before the Leds' I mean I a 470 Ohm resistor between the level shifter and the first Led. I haven't tried a capacitor before the Leds, but I don't think the power supply is the problem.

The problem occurs all over the matrix, so it is not limited to certain columns. However, the later rows are surely more affected more than the first. I do use Bernd's wiring.

I think I have to rewire everything. I'll use Neopixels (3 separate clock/data) instead, and no clips.

Thanks a lot. Regads, Arjan

bk1285, 2018-01-18:
@Arjan444 
if the issue reappears, feel free to reopen at any time.

closed 2018-01-18 by bk1285

Arjan444, 2018-01-18:
Thanks Bernd. Stable since I replaced the leds with Neopixels.