bk1285/rpi_wordclock #75: Another ways of control...

Javi-es, 2017-12-14:
Hello for all

Is possible to simulate the three buttons with a bash script or sending a JSON commands with curl? (I use the web_interface)

in example, with cron to change the brightness at night/day, or change the colors if the weather changes, start a ip plugin on new eth0 connection, etc..

Thanks a lot.

bk1285, 2017-12-20:
Yes, it is possible. I'll setup some docs on this...

Best,
Bernd

bk1285, 2018-04-25:
Hi @Javi-es,
FYI: There's a branch for a remote control via webinterface within this repository. It uses JSON, while still beeing highly experimental....

bk1285, 2018-04-25:
... here the branch: https://github.com/bk1285/rpi_wordclock/tree/feature/webinterface

bk1285, 2018-04-25:
... when running it, you can access the wordclock via http:
http://192.168.X.Y
http://192.168.X.Y/list
http://192.168.X.Y/active

Using POST, you can also set the current plugin: http://192.168.X.Y/api

FrankX0, 2018-05-27:
When will you release this functionality?
Very useful!

One issue to resolve though: some (e.g. Tetris, Sunrise, Time as words) plugins are executed again when switching to a new plugin.
So:

- Select plugin Tetris
- Plugin is executed, wait for it to finish
- Select a different plugin
- Tetris is executed again, followed by the execution of the newly selected plugin

Could not yet find why this happens.

bk1285, 2018-05-27:
Hi @FrankX0, the issue, which you mentioned is the reason not to release it yet. Hope to fix this soon.
@mrksngl: time for another wordclock session... :D

bk1285, 2018-06-10:
Hi @FrankX0,

checkout the latest develop: https://github.com/bk1285/rpi_wordclock/tree/develop
I improved the webinterface and event handling, which should work now without the bug, described above.

I'm happy to receive feedback in order to merge to master one day... ;)

Best,
Bernd

FrankX0, 2018-06-10:
Hi Bernd.

I will give it a try, thanks!

One related question.
I would like to be able to set the colour through the web interface.
How can I change the value of `word_color` in the time_default plugin from web_interface.py?

FrankX0, 2018-06-16:
Hi Bernd,

I gave it a try: works fine!
One remark: I added a `return("OK")` at the end of `@web_interface.app.route('/api', methods=['POST'])`, to prevent errors (_View function did not return a response_).

bk1285, 2018-06-16:
That's great news, @FrankXO!

I plan to provide an interface for setting colors, as well.... ;)

Best,
Bernd

FrankX0, 2018-06-17:
That would be great!
I am currently working on a new front-end, using your web-interface.
I can currently select the plugin and have the buttons available for setting the colors and playing Tetris :-) For that I extended your web interface to also be able to introduce button actions.
Can you extend the web interface such that we can freely set the color (RGB) of each clock element separately (words, minutes, background)?

bk1285, 2018-09-16:
Hi @FrankX0 

checkout -- once again -- latest develop: I just released a new version of the wordclock-api with the option to set colors as well as some documentation.

Best,
Bernd

FrankX0, 2018-09-17:
Hi Bernd. Thanks!
I will certainly give it a try.

FrankX0, 2018-09-20:
Hi Bernd.

I now have some hand-on experience with the new API: GREAT!
I had to do some porting for my clock control web interface. For that the API documentation worked really well!
Some feedback:
- I found out that I had to use the [develop](https://github.com/jgarff/rpi_ws281x/tree/dev) version of the NeoPixel library.
- With this new implementation the red and green colours are swapped, resulting in a very red matrix clock-face :-)

A preview of the web interface:
[image]
With the current version I can select the plugin, use the control buttons, set some colours and the brightness.
If I'm a bit further I would like to integrate in your source.

Regards,
Frank.

bk1285, 2018-09-21:
Hey @FrankX0,

that's great news! The web interface looks really nice!!

If I find the time, I'll expand the API to set the color elementwise (words, minutes, background).

Best,
Bernd

FrankX0, 2018-09-21:
Hi Bernd,

What about the swapping of green and red I am facing?
I thought it has to do with the strip type (`ws.WS2811_STRIP_GRB`) set in wordclock_display.py.
But changing that to `ws.WS2811_STRIP_RGB` did not change the behaviour.

Any ideas?

bk1285, 2018-10-12:
> Can you extend the web interface such that we can freely set the color (RGB) of each clock element separately (words, minutes, background)?

Hi @FrankX0, I just pushed API verison 4.1 to optionally support specific color settings for words, minutes and background as well.

Best,
Bernd

FrankX0, 2018-10-13:
Hi Bernd, Thanks for the update!
I will certainly have a look and integrate this into the graphical interface.

Thanks again,
Frank.

FrankX0, 2018-10-16:
Hi Bernd,

I made an initial commit (#98) for the graphical web interface.
Please give it a try. All feedback is welcome, including my (lack of) coding style.
I expect to still update it in the coming period.

Mit freundliche grüße,
Frank.

bk1285, 2021-07-24:
Hey all, I guess this resolved/outdated? @FrankX0

Best,
Bernd

closed 2021-07-24 by bk1285