bk1285/rpi_wordclock #77: Broken Amazon Links

ebarlas, 2018-01-01:
I'm not able to view the Amazon links. Any suggestions?

[image]

bk1285, 2018-01-03:
Indeed... :/

However, to a certain extend: 3rd party issue without any chance to get the page back, I fear.... 

So you need a list of components, right?

Best,
Bernd

ebarlas, 2018-01-03:
Yes, that's right. The list under general considerations is excellent, but specific parts would be very useful as well.

BjornHoorelbeke, 2018-03-15:
Great project!! I would also be interested in which components to order, can you please provide a list?

bk1285, 2018-04-25:
Here we go...:

- The LEDs: https://www.ebay.de/itm/5m-150-RGB-LED-Strip-Stripe-schwarz-mit-WS2812B-5050-SMD-WS2812-30-LEDs-m-T9-/122407573083?hash=item1c800f565b
  - Ensure, that the spacing fits your clock size (letter spacing)
- A Raspberry Pi + SD-card
  - E.g. RPi0 with Wifi is a nice choice
- A power supply: https://www.amazon.de/gp/product/B01NAN9QUD/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1
- Optionally: Some buttons: https://www.amazon.de/gp/product/B00JFTAIQS/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1

closed 2018-04-25 by bk1285