bk1285/rpi_wordclock #78: Support for multiple stencils, operated by servos

ruedli, 2018-01-02:
My clock has three moveable stencils, that can be moved with servo's.

All is working now, and I have created a plugin to allow for choosing the template to use for time_default, sunrise and matrix_time.

You need to install and enable i2c support and import the ADAfruit_PCA9685 module.

I have:
- Added my wiring
- Added a stencil "dutch2"
- Added a stencil "english2"
- Added a plugin "stencil_changer"
- updated the configuration file to allow configuring the servos to use and their settings for activating the different stencils
- updated "woordclock.display.py", so that it can also be used for changing stencils in other plugins.
- updated the sunrise plugin, so that it uses "dots" for animation and the correct (selectable) template for showing the time for sunrise/sunset.
- moved the time_xxxxx.py files from the time_default plugin to the wordclock_tools directory and made the selection of the correct time translation generic: until now every plugion had to use their own selection of time_xxxx modules, and they were not up-to-date, nor complete.

The code and configuration files contain explanation and examples.

I have tested, that this also works "without" the new hardware for the servo's, so it does not break the functionality for clocks without servos. 

I have everything versioned in git, but have trouble committing this delta (I do not see "files" on a branch to commit to), so for now enclose the "unified diff" as pdf.

[unfied diff Ruud Rademaker inplementation of servo changer.pdf]
(https://github.com/bk1285/rpi_wordclock/files/1597659/unfied.diff.Ruud.Rademaker.inplementation.of.servo.changer.pdf)

as well as the modified / moved files.

[rpi_wordclock.zip](https://github.com/bk1285/rpi_wordclock/files/1597674/rpi_wordclock.zip)

Please let me know how I can contribute it into the repository, and what to do next. Will try to upload a video as well with this new "feature" all people that have seen it so far, were very impressed... And, obviously the clock pleases me as well. (Perhaps exchange by mail further info, that would be much more efficient)

Thnx Bernd for creating the excellent starting conditions for having this feature!!!! As I implemented these changes, I got very impressed with the nice framework you have created, software!

ruedli, 2018-01-02:
And... here is the video....

https://youtu.be/XTHH30S8Aao

With some pictures of the mechnism, Software, with a new plugin, is now fully functional, I intend to build a second row of 3 servos to the right, currently I only use 3 servos, one for each stencil.

[image]

[image]

This picture shows the driver for the up to 16 servos, a 2 dollar TDA-9685 breakout. (Uses i2c and does all timing stuff, without bothering the raspberry pi)

[image]

bk1285, 2018-01-03:
Hi @ruedli, really nice clock! Thanks for sharing!!

....definitely a strong candidate for the gallery (...to come...)

ruedli, 2018-01-04:
Hi Bernd,

As mentioned, I have the code managed under git here on my PC, how do you
want to proceed now?

Leave it on my PC only?
-or-
Review the changes and merge it with your code repository?

I set it up in such a manner, that it does not break the functions for
people not having servos to operate the stencils. At the same time, people
who swap their stencil manually, with magnets e.g. could benefit from the
"stencil changer" plugin.

There are new dependencies however: for i2c and the Adafruit library, they
must be included in the software installation doc.

You can mail me to my private mail, to exchange ideas on the merge, also I
could program other new feature, I intend a few more addons myself.

cheers Ruud

(ps. I also resolved a "bug" when you configure the initial brightness
level at a vaule larger then 247, it crashes the clock in an attempt to set
the intensity at values bigger then 255, the "check" for bigger then 255
was at the wrong place, and the "change direction logic not working
properly)

bk1285, 2018-01-04:
Hi @ruedli,

please create a [pull request](https://help.github.com/articles/about-pull-requests/) to "develop".

Then we can discuss changes in detail and decide, whether to integrate the soures (and documentation) into develop (preferred!) or leave it on a separate branch.

Best,
Bernd

ruedli, 2018-01-04:
Hi Bernd,

Sorry I feel like an idiot, but despite more then 15 years experience
selling configuration management solutions, I do not succeed in uploading
the delta's to the files...

When I create the pull request and "drag-and-drop'.py files, I get a file
type not supported. I never see a directory tree, where I would expect to
be able to upload individual files.

Finally, I did not base my development on "develop",  but on the master
copy of a week ago. The develop already contains a change. Even if you
already migrated it, adding my change there should create either a conflict
or undo the intended change currectly in the development branch. (If I
understand correctly how this is supposed to work...

So please, a little help on making the pull request work for us, so that I
can update and add .py files and that we use the correct baseline for the
pull request...

(sorry for not understanding)