bk1285/rpi_wordclock #81: I've wired my leds false way

dennisj03, 2018-02-02:
Hello, i wired my leds the False way, and now the " IT IS" is not in the upper left corner it is in the upper right corner. Can i change the code so that it fits my wiring or do i need to change everything?

Ch4r0s, 2018-02-02:
Hi,
I've also wired my clock a different way (but I wanted so^^). You just have to add a "wiring" class in "./wordclock_tools/wiring.py"

If you like to, you can show me a pic of your wiring and I create your snippet.

greatz

dennisj03, 2018-02-05:
Hello,
My wiring looks like this:

[image]

bk1285, 2018-02-05:
Front or back-view? "It is...." is top right or top left?

dennisj03, 2018-02-05:
this is the view from the back. The Raspberry pi is top left

Ch4r0s, 2018-02-06:
Try this:
add to your wordclock_tools/wiring.py:
at top in the "if-elfif-cases":

```
elif wiring_layout == 'your_wiring':
    self.wcl = your_wiring(self.WCA_WIDTH, self.WCA_HEIGHT)
```

and as new class:
```
class your_wiring:
    def __init__(self, WCA_WIDTH, WCA_HEIGHT):
        self.WCA_WIDTH   = WCA_WIDTH
        self.WCA_HEIGHT  = WCA_HEIGHT
        self.LED_COUNT   = self.WCA_WIDTH*self.WCA_HEIGHT+4

    def getStripIndexFrom2D(self, x, y):
        if y%2 == 0:
            pos = (self.WCA_HEIGHT*y)+x+2
        else:
            pos = (self.WCA_HEIGHT*(y+1))-x-1
        return pos

    def mapMinutes(self, min):
        if min == 1:
            return 1
        elif min == 2:
            return 0
        elif min == 3:
            return self.LED_COUNT-1
        elif min == 4:
            return self.LED_COUNT-2
        else:
            print('WARNING: Out of range, when mapping minutes...')
            print(min)
            return 0
```
can't test it right now, pls give feedback

dennisj03, 2018-02-10:
hello,
i am not able to test this for another 1 and a halv week, but i will answer soon

FrankX0, 2018-02-21:
Somewhat related to this question: is it possible to leave out the minute LEDs?

bk1285, 2018-04-25:
Hi @FrankX0,
in which sense? Having them available, but switched off? ... or even not having the leds built-in? 

From the software, they are addressed here: https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_tools/wordclock_display.py#L191

Best,
Bernd

FrankX0, 2018-04-26:
Thanks Bernd!
I was planning a clock face without the minute LEDs.
But ended up anyway with them in.
I am building a smaller version (22 x 22 cm) with 60 LEDs/m strings.
Looking forward to get it to work in the coming days.
Thanks for this nice project!

Bluetex07, 2018-04-27:
Did you test the code? I also have the problem that Es ist is Now displayed mirrored in the Upper Left corner.

I used bernds wiring but forgot that i have to mirror this as im using it on the back which will be mirrored onto the MDF Plate

webdisaster, 2018-08-17:
Have a diferent wiring, so signs can use bigger front 
[image] 4 Leds are at the end of the string. Leds are not mirrored because they are on a pcb. had not found out ,how your programming has done so how can i  make a own wiring file?

bk1285, 2018-11-05:
Hi @webdisaster,

is the image of the wiring seen from the front or the back of the clock?

Best,
Bernd

webdisaster, 2018-11-07:
this is front view

thanks
bernhard

webdisaster, 2018-11-07:
4x minute  leds are at the end of the led string

closed 2018-11-07 by bk1285

webdisaster, 2019-12-29:
Hi 
Tried it a bit late but it works perfect!