bk1285/rpi_wordclock #82: Wordclock without changing colors

spintike, 2018-02-07:
Hello Bernd!
[image]

I have successfully built a wordclock on my own - except for the excellent software you've provided!

Now to be honest I'm a bit unsatisfied with my own wordclock changing the colors (as you can see on the attached pictures). The question at hand would be if there's a slick way to deactivate the coloring at all?

Until I was not able to understand the code to that point where I can deactivate it.

Help is greatly appreciated!

bk1285, 2018-02-07:
Hi @spintike

Nice build! Do you have buttons attached to the clock?

Please check your wordclock config for this setting:
[wordclock_interface]
type = gpio_low

... and try with set to "gpio_high"

Best,
Bernd

bk1285, 2018-02-07:
PS: Hope you don't mind, if I add your swabian clock to the "Hall of fame": https://www.pinterest.de/berndkrolla/wordclock-gallery/
;)

spintike, 2018-02-08:
That's fine for me @bk1285 :-)

Unfortunately this is not what I meant with my original post.

Right now I have no buttons attached to the clock. It's just that it's changing the LED colors from time to time and I want to get rid of it - I want to state that its a fine piece of code to achieve this function but in my case it's just not needed...

bk1285, 2018-02-08:
Hi,

so if you have no buttons attached, you need to set 

[wordclock_interface]
type = gpio_high

Please let me know, if this does not resolve the issue. In this case it's probably a major issue......

spintike, 2018-02-15:
Hi again!

Sorry for answering so late!

The GPIO setting did the trick, now everything is as wanted 👍

closed 2018-02-15 by spintike