bk1285/rpi_wordclock #83: Software Problem

Bluetex07, 2018-02-13:
Is there any more detailed part of what and how i have to install this to the rasp?

Tried it but i can´t figure out what files i need, is there a chance you could give a short summary?

spintike, 2018-02-15:
Hello @Bluetex07!

If I may comment on this in a short manner: I assume that you're having the RPi already set up and working right?

Please have a look on the [software setup page](http://rpi-wordclock.readthedocs.io/en/latest/doc_software_setup.html). There you'll find all the things needed for installing the wordclock software on your RPi.

Bluetex07, 2018-02-15:
I understand the Setup, but there was also a link to download the clock files with many folders in it.Do i need this or is it the same like the Command to download clock Software?

spintike, 2018-02-16:
@Bluetex07 please specify what you mean by that.

TSTr06, 2018-03-02:
Hey there,
first of all thanks for this great software! 
I built a wordclock with bernds wiring and the german version is working fine.
Now I want to change the language to swabian with the german2 stencil.
But the according changes in the config file have no effect on the shining LEDs and I always get the error: "Could not detect german2 / swabian". Instead the german version gets automatically chosen.
What is my mistake? Do I have to adapt other files to get the swabian time running?
Thanks a lot!

bk1285, 2018-03-02:
Hi,
please ensure the correct wording in the config file: german2 OR swabian.
Check here for the available options/spellings: https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_plugins/time_default/plugin.py#L39

Best,
Bernd

TSTr06, 2018-03-02:
Hey,
thanks for your answer. It's clear that I can choose only one language. What I wanted to say is that neither german2 NOR swabian is working. Instead I get the error "Could not detect swabian" respectively "Could not detect german2" and in the next line "Choosing default german".

My next question concerns the stencils. Is every german language (especially german, german2, swabian) compatible with the stencils german and german2? Or is for example the combination stencil: german2 and language: swabian not working?

Sorry for so many questions, this is my first project with an rpi 🔨 

Thanks!

bk1285, 2018-03-02:
@TSTr06: Let's discuss in #86 since it's indeed a bug (which is unrelated to the request, opened by Bluetex07)

closed 2018-04-25 by bk1285