bk1285/rpi_wordclock #84: question regarding german "quarter to x" or "three quarters x"

spintike, 2018-02-15:
Hello!

This part is quite tricky to explain in english but I'll try anyway.

There are regions in Germany where it's common sense to say "it's quarter before six" for 17:45. And then there are regions where they say "three quarter six" which is essentially the same.
The original german layout of the wordclock however is made to meet the first saying. This also means that the original layout contains the word "ZWANZIG" ("twenty") in order to achieve the time for, let's say "ZWANZIG NACH ACHT" ("twenty past eight") or "ZWANZIG VOR NEUN" ("twenty minutes before nine").
Now in my own swabian layout there's no "ZWANZIG" available...due to reasons...I thought it'll be good to replace it with "ZEHN VOR HALB" ("ten before half") or "ZEHN NACH HALB" ("ten after half") which is essentially the same. After this stupid decision I realized that my wordclock is showing the wrong timestamps.
Let's assume it's 08:20 right now and from my POV this should be shown as "ZEHN VOR HALB NEUN" ("ten minutes before half to nine", sounds strange but it is that way in german). Unfortunately it now shows "ZEHN VOR HALB ACHT" ("ten minutes before half to eight"), I assume because the hours mapping is wrong.

Long story short, can somebody advise which file I have to review in order to fix this? It'll be awesome to get this thing working like I want to...

Thanks in advance!!!

bk1285, 2018-02-15:
Hey @spintike,

within the config file you can specify, which time the time_default plugin loads. According to the config, you have to pick the corresponding file to edit here: https://github.com/bk1285/rpi_wordclock/tree/master/wordclock_plugins/time_default

Best,
Bernd

spintike, 2018-02-17:
YES! It's working as I want to have it 👍 thanks @bk1285!

bk1285, 2018-02-19:
Hi @spintike,
was the issue related to your stancil layout or is it a software bug? In the latter case, I'd be great, if you'd provide some details, how you fixed it.
Best,
Bernd

spintike, 2018-02-20:
It was only a configuration issue @bk1285.

I just have to change the formula where the program calculates the minutes in [https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_plugins/time_default/time_german.py](url), line 39. The original setting '4' meant that only when it's twenty past an hour it shows the next hour...I changed it to '2' and afterwards it's working fine :+1:

closed 2018-02-21 by spintike