bk1285/rpi_wordclock #85: DMA 5 Problem with SD Card on Pi Zero and Pi3

thejoe8495, 2018-02-19:
There is an SDCard issue on the new Pis. The DMA5 is used to access the SDCard when you start the Wordclock the pi will eject the SD Card and crash.
Please change
https://github.com/bk1285/rpi_wordclock/blob/f61f8f3c674cd7c9393905a19ed8d397ecd7ca9d/wordclock_tools/wiring.py#L22
to DMA 10

For more information:
https://github.com/jgarff/rpi_ws281x/issues/188

bk1285, 2018-02-19:
Hi @thejoe8495 
excellent! Thanks for reporting! I already ran into the issue, but wasn't able to track it down...
Best,
Bernd

bk1285, 2018-02-19:
@thejoe8495 can you provide feedback, if it works with the latest master?

thejoe8495, 2018-02-19:
Yes it work on my pi zero, for the pi3 i need to wait my octopi finished with printing :wink:

bk1285, 2018-02-24:
Please reopen, in case it does not work with the Pi3.

Best,
Bernd

closed 2018-02-24 by bk1285