bk1285/rpi_wordclock #86: Fails to select language according to config file

bk1285, 2018-03-02:
Moved to new issue: #83  @TSTr06 

Hey,
thanks for your answer. It's clear that I can choose only one language. What I wanted to say is that neither german2 NOR swabian is working. Instead I get the error "Could not detect swabian" respectively "Could not detect german2" and in the next line "Choosing default german".

My next question concerns the stencils. Is every german language (especially german, german2, swabian) compatible with the stencils german and german2? Or is for example the combination stencil: german2 and language: swabian not working?

Sorry for so many questions, this is my first project with an rpi hammer

Thanks!

bk1285, 2018-03-03:
Hi @TSTr06,
can you check, if the latest develop branch fixes your issue?

Best,
Bernd

TSTr06, 2018-03-07:
Hey bk1285,

I tried your proposal and now I get this error:

Failed to import plugin time_default!
Could not detect language: swabian.
Choosing default: german
Imported plugin 6: "sunrise".
Imported plugin 7: "ip_address".
Imported plugin 8: "shutdown".
Imported plugin 9: "feed_parser".
Imported plugin 10: "restart".
Setting up wordclock socket
Traceback (most recent call last):
  File "wordclock.py", line 166, in <module>
    word_clock.run()
  File "wordclock.py", line 128, in run
    self.run_next_index = self.default_plugin
AttributeError: wordclock instance has no attribute 'default_plugin'

bk1285, 2018-03-11:
Strange.... Unfortunately I could not reproduce it after the provided fix: Works now on my clock....

bk1285, 2018-03-12:
Which python version are you running? Works on my Pi with python 2.7.3

TSTr06, 2018-03-13:
I'm using python 2.7.13

plotaBot, 2018-12-25:
I also ran into this problem since I am using german2. Unfortunately this only occurs in the log while the clock is displaying the right language. Though it is a weird bug
Are you using the developer mode? If so lease keep in mind that you have to change the string inside wordclock_tools/GTKstrip in line 27 according to your desired template. Have a look at [my fork](https://github.com/plotaBot/rpi_wordclock/blob/master/wordclock_tools/GTKstrip.py#L27) where I changed it to german2.

Please also consider taking a look at my pull request #111 where I also changed the language selection for matrix_with_time and sunrise to the more convenient version at which it is choosen by the one set in time_default. Also the corresponding languages are imported. @bk1285 this is a little maintenance isssue;) 

PS: While writing this I discovered the cause of this issue: The plugin sunrise selects the language from plugin_time_default. The selection fails because the only languages available to it are german, dutch and swiss_german. It wil be fixed with my pull request;)

plotaBot, 2018-12-27:
As soon as #111 is merged, you should be able to test if this behaviour still occurs

plotaBot, 2018-12-27:
Maybe my #111 is unrelated to the original problem ^^
So the problem is not occuring when I am using the clock

bk1285, 2018-12-29:
Hey! Checkout current master. Is it fixed there, or does is still occur?

sarosr, 2019-05-29:
Hello bk1285, i try the master branch and also the develop, but there is the same error:

_Interface type set to gpio_low
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
  Wiring layout: bernds_wiring
  Setting language to italian.
Could not detect language: italian.
Choosing default: german_

If I choose english, it work

Interface type set to gpio_low
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
  Wiring layout: bernds_wiring
  Setting language to english.

thank you

plotaBot, 2019-05-30:
It appears that the language italian is not configured as a time_italian.py **yet**.

Checkout the [configurator](https://files.qlocktwo.com/Q2/embed.html?color=BLACK%20ICE%20TEA&controls=true&language=IT&size=450) page from the orignal wordclock and if you save the page as a local html you are able to reverse engineer the original time display! Make sure you change your computers clock to speed up that process;)

@bk1285 please consider adding this [configurator](https://files.qlocktwo.com/Q2/config.html) to the docs or else for people who like to add a new language.

sarosr, 2019-06-29:
Hello, i don't know how upload the file, so i past here the code of time_italian.py

----------------------------------------------------------------------------------------------------------------
""" Provided by Saruzzo """

class time_italian:
    """
    This class returns a given time as a range of LED-indices.
    Illuminating these LEDs represents the current time on an english WCA
    """

    def __init__(self):
        #self.prefix = range(0,2) +  range(3,5) # -> IT IS
	self.prefix = range(0,4) +  range(5,7) # -> SONO LE 

        self.minutes=[[], \
            # -> FIVE PAST e cinque
            #range(28,32) + range(44,48), \
	    range(77,78) + range(93,99), \
            # -> TEN PAST e dieci
            #range(38,41) + range(44,48), \
	    range(77,78) + range(99,104), \
	    # -> QUARTER PAST e un quarto
            #range(13,20) + range(44,48), \
	    range (77,78) + range (79,81) + range(82,88), \
	    # -> TWENTY PAST e venti
            #range(22,28) + range(44,48), \
	    range(77,78) + range(88,93), \
            # -> TWENTYFIVE PAST e venticinque
            #range(22,32) + range(44,48),  \
	    range (77,78) + range(88,99), \
	    # -> HALF PAST e mezzo
            #range(33,37) + range(44,48), \
            range (77,78) + range (105,110), \
            # -> TWENTYFIVE TO meno venticinque
            range(73,77) + range(88,99), \
            # -> TWENTY TO meno venti
            #range(22,28) + range(42,44), \
	    range(73,77) + range (88,93), \
	    # -> QUARTER TO meno un quarto
            #range(13,20) + range(42,44), \
	    range(73,77) + range(79,81) + range(82,88), \
	    # -> TEN TO meno dieci
            #range(38,41) + range(42,44), \
	    range(73,77) + range(99,104), \
	    # -> FIVE TO meno cinque
            #range(28,32) + range(42,44) ]
	    range(73,77) + range(93,99) ]
	    # -> TWELVE dodici
        self.hours= [range(8,11) + range(44,50), \
            # -> ONE l'una
            range(11,12) + range (13,17), \
            # -> TWO due
            range(8,11) + range(18,21), \
            # -> THREE 3
            #range(61,66), \
	    range(8,11) + range(22,25), \
            # -> FOUR 4
            #range(66,70), \
	    range (8,11) + range(55,62), \
            # -> FIVE cinque
            range(8,11) + range(66,72), \
            # -> SIX 6
            range(8,11) + range(63,66), \
            # -> SEVEN 7
            range(8,11) + range(50,55), \
            # -> EIGHT 8
            range(8,11) + range(25,29), \
            # -> NINE 9
            range(8,11) + range(29,33), \
            # -> TEN 10
            range(8,11) + range(33,38),\
            # -> ELEVEN undici
            #range(82,88), \
	    range(8,11) + range(38,44), \
	    # -> TWELVE dodici
            #range(93,99)]
	    range(8,11) + range(44,50)]
        # -> l'una
        self.full_hour= range(11,12) + range (13,17)

    def get_time(self, time, purist):
        hour=time.hour % 12+(1 if time.minute/5 >= 7 else 0)
        minute=time.minute/5
        # Assemble indices
        return  \
            (self.prefix if (not purist and hour !=1) else []) + \
            self.minutes[minute] + \
            self.hours[hour] + \
            (self.full_hour if (hour == 1) else [])

---------------------------------------------------------------------------------------------------------

thank you, bye

bk1285, 2019-08-01:
Hi @sarosr,

thank you for sharing! Does the stancil layout correspond to the one, already contained within this repository? E.g. https://github.com/bk1285/rpi_wordclock/tree/master/wordclock_layouts/italian_11x10
Especially https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_layouts/italian_11x10/stancil_front.png

Best,
Bernd

sarosr, 2019-08-04:
yes, i use the italian 11x10 stencil_front.png

thank you for your great work.

bk1285, 2021-07-24:
Please reopen, if not resolved.

closed 2021-07-24 by bk1285