bk1285/rpi_wordclock #87: Wordclock lost the 11 letters

pizzajt, 2018-03-11:
First: Many thanks to the Great work and a wonderful wordclock!

It Looks great.

Since the last days my wordclock lost the last 11 letters and the last 2 dots in the Strippe.

So the word „ES“ is missing.
I used Strandtest, it stops after „GRE“

The clock worked Fine the last year an I didn‘t change anything.

pizzajt, 2018-03-11:
https://photos.app.goo.gl/nZc4WdQaYKL7z3ah2

bk1285, 2018-03-11:
Hey @pizzajt,
I'd guess, it's a hardware issue: Check the soldering at the point, where the 11 letters "get lost".
Possibly, you have to renew the solderimg or replace some defect LEDs at that point.

Best,
Bernd

pizzajt, 2018-03-11:
Hi Bernd.

Are you shure this will be a hardware issue?

At this Point there is no soldering, i use 1 strip per coloumn.

So i have to cut the strip and will change the led „S“.

Many thanks!

pizzajt, 2018-03-11:
I changed the LED - everything is working fine!
Here you can find some pictures of my „making of“:

https://photos.app.goo.gl/oGf6EfMomIlovqdL2

pizzajt, 2018-03-11:
I ordered the Front Panel at

LTO GmbH
eMail: info@lto-gmbh.de

Frontplatte
Typ: DXF-Import
Material: Stahl DC01
Materialdicke: 1mm
Oberfläche: matt 
Datei: wiring_front_JT7.dxf
Interne Datei: dxf_57e11e69d6308.dxf

23,94 €

bk1285, 2018-03-11:
Great, good to know!
...and thanks for the feedback. I'll add the images to the wordclock gallery on pinterest, if you don't mind... ;)

closed 2018-03-11 by bk1285