bk1285/rpi_wordclock #88: Append "S" to EIN (in case of "EIN UHR") - How to deactivate

spintike, 2018-04-16:
Hello @bk1285 

I'm struggling with the S being appended to EIN when the wordclock is showing a time where one ("EIN") is occuring.
I already deactivated the lines in time_as_words_german.py corresponding to the appending of "S" but after saving the file it's still showing on my watch.

Problem ist that I'm using a custom layout (see screenshot) and the character which was originally assigend to the "S" in question is now glowing in the middle of the wordclock :(

Thanks in advance!
[image]

bk1285, 2018-04-25:
Meanwhile any progress? I see the "S" and the "H" to be problematic?

However, a swabian layout is already available here: https://github.com/bk1285/rpi_wordclock/blob/master/wordclock_plugins/time_default/time_swabian.py

... you can select it within the wordclock_config.cfg.

Maybe that helps?

Best,
Bernd

bk1285, 2018-11-01:
Still an issue? Reopen, if this is the case...

closed 2018-11-01 by bk1285

spintike, 2018-11-01:
Whoops, forgot to post...issue is solved so nö need to reopen the case :)

Thanks.