bk1285/rpi_wordclock #91: Import Error: No module named neopixel

schweizschweiz, 2018-05-14:
Hi There,

I just downloaded the software and then tried to start the wordclock I get an Error Messsage (see the Attachement). 
I already formatted my sd card and tried it again. Same error message. 
it's confusing cuz i already had the software running until i decided to reinstall it because i sometimes got an error and the clock stoped.

did something change the software? Is it possible to download the missing neopixel import somewhere else

best regards
Christian
https://photos.google.com/photo/AF1QipNoWMoyoFN5lERQnEbHqk8XIP5dYt4GXjOVpuPJ

schweizschweiz, 2018-05-14:
i think the mistake happens when i enter

`git clone https://github.com/jgarff/rpi_ws281x.git`

maybe the libary has been updatet?

`cd rpi_ws281x`
`sudo scons`
`cd ~/rpi_ws281x/python`
And then when i enter `sudo python setup.py install` it says 

> File is not a zip file

as Herr Krolla mentioned in his manual it's possible to use a libary which is more likely to work

> https://github.com/jgarff/rpi_ws281x/tree/2f9e03c45ba0964029204db565ad9d6233b3a1a6

i dont understand what kind of command i have to hit in the cmd line to refer to this special link

i tried just `git clone https://github.com/jgarff/rpi_ws281x/tree/2f9e03c45ba0964029204db565ad9d6233b3a1a6` but it didn't work. Unfortunately i do not know a lot about coding.

Thanks for any help

schweizschweiz, 2018-05-15:
okay never mind. I did find a solution
for anyone that may have the same Problem here's what I did.
I thougt that maybe the libary was not installed correctly (obviously).  I deleted the whole driectory 

> rpi_ws281x
with `rm -rf rpi_ws281x/` and then followed Herr Krollas manual from the beginning until `git clone https://github.com/jgarff/rpi_ws281x.git`. 
then i used `sudo pip install rpi_ws281x` to install it properly.
and then just follow the manual to the end.

This can be closed now

closed 2018-05-15 by schweizschweiz

Djangx, 2021-03-21:
Hi. I have the same issue, and have tried your solution, but it did not work for me. Any ideas what I can try? Thanks in advance.

phenze, 2021-03-22:
Take a look here. Try updating to python3.

https://github.com/bk1285/rpi_wordclock/issues/167